espressif/idf-eclipse-plugin #502: Deleting project cause espressif IDE to stop responding

tsctrl, 2022-03-06:
**Describe the bug**
I getting issue several time where the IDE was not responding when deleting the project from workspace.
I cant stop it, Kills the process in windows but the ide did not terminated either and i have to restart my pc. 
i am guessing this is related to espressif ide try to cleanup file created during the build.

**To Reproduce**
Steps to reproduce the behavior:
1. import some medium size project
2. build & Flash
3. delete project
4. See error

[image]

Environment Information:
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

ESP-IDF Tools Information:
Help > Espressif Tools Manager > List Installed Tools
idf_tools.py list

xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
esp-2021r2-8.4.0 (recommended, installed)
esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
cmake: CMake build system
3.16.4 (recommended, installed)
openocd-esp32: OpenOCD for ESP32
v0.10.0-esp32-20211111 (recommended, installed)
ninja: Ninja build system
1.10.2 (recommended, installed)
idf-exe: IDF wrapper tool for Windows
1.0.1 (recommended, installed)
ccache: Ccache (compiler cache)
3.7 (recommended, installed)
dfu-util: dfu-util (Device Firmware Upgrade Utilities)
0.9 (recommended, installed)
IDF Eclipse Plugin Version:
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
ESP-IDF Eclipse Plugin
Version: 2.4.0.202201280731

Eclipse Error log:
No error message

Thank you!

kolipakakondal, 2022-04-04:
Hi @tsctrl Sorry for the delay in responding. If you still have the same problem. Could you please provide us with the stack trace log to investigate further?

https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-report-a-deadlock-or-eclipse-hang

tsctrl, 2022-04-07:
[image]

still reproducible.. deleting project took so much time and hang. i have to kill eclipse process
try import and compile medium size project. and delete the project from workspace

kolipakakondal, 2022-04-07:
Hello @tsctrl sorry for the inconvenience. To troubleshoot further could share the java stack info which I have requested earlier.

kolipakakondal, 2022-09-30:
Hello, If you still have the problem, please provide stack details as requested earlier and reopen the ticket to investigate further. thanks

closed 2022-09-30 by kolipakakondal