espressif/idf-eclipse-plugin #503: Switching between 4.3.2 and 4.4.0 idf will cause project build to failed

tsctrl, 2022-03-06:
**Describe the bug**
Switching from different idf version causing project build failed. Cause by the Python interpreter used by the project was not switch automatically to v4.4.0

**To Reproduce**
Steps to reproduce the behavior:
1. Create blink project using IDF version 4.3.2
2. Build and flash
3. Switch Espressif IDF to version 4.4.0 using menu Espressif->Download or use ESP-IDF and use existing directory 4.4.0
4. Build and See error as below

**Expected behavior**
Able to build successfully

Launch configuration is correct.
[image]

Environment Information:
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

ESP-IDF Tools Information:
Help > Espressif Tools Manager > List Installed Tools
idf_tools.py list

xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
esp-2021r2-8.4.0 (recommended, installed)
riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
esp-2021r2-8.4.0 (recommended, installed)
esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
2.28.51-esp-20191205 (recommended, installed)
cmake: CMake build system
3.16.4 (recommended, installed)
openocd-esp32: OpenOCD for ESP32
v0.10.0-esp32-20211111 (recommended, installed)
ninja: Ninja build system
1.10.2 (recommended, installed)
idf-exe: IDF wrapper tool for Windows
1.0.1 (recommended, installed)
ccache: Ccache (compiler cache)
3.7 (recommended, installed)
dfu-util: dfu-util (Device Firmware Upgrade Utilities)
0.9 (recommended, installed)
IDF Eclipse Plugin Version:
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
ESP-IDF Eclipse Plugin
Version: 2.4.0.202201280731

Eclipse Error log:
`
Diagnostic information:
    IDF_PYTHON_ENV_PATH: C:\Users\.espressif\python_env\idf4.4_py3.8_env
    Python interpreter used: C:\Users\.espressif\python_env\idf4.3_py3.9_env\Scripts\python.exe
    Warning: python interpreter not running from IDF_PYTHON_ENV_PATH
    PATH: C:\Users\.espressif\tools\xtensa-esp32-elf\esp-2021r2-8.4.0\xtensa-esp32-elf\bin;C:\Users\.espressif\python_env\idf4.3_py3.9_env\Scripts;C:\esp-idf\esp-idf-v4.3.2\tools;C:\Users\AppData\Local\Programs\Python\Python39\;
.....
CMake Error at C:/Development/esp-idf/esp-idf-v4.4.0/tools/cmake/build.cmake:286 (message):
  Some Python dependencies must be installed.  Check above message for
  details.
Call Stack (most recent call first):
  C:/Development/esp-idf/esp-idf-v4.4.0/tools/cmake/build.cmake:415 (__build_check_python)
  C:/Development/esp-idf/esp-idf-v4.4.0/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:6 (project)
`

Thank you!

kolipakakondal, 2022-03-09:
Hi @tsctrl Thanks for reporting the issue. I do believe this is a good feature to have where each project has its own configuration settings based on the IDF Version. Unfortunately, this is not supported yet.

The workaround would be - when you switch IDF version, please run `Espressif > ESP-IDF Tools Manager > Install Tools` command and this will change the IDF version-specific settings that include python interpreter, openOCD, etc.

kolipakakondal, 2022-08-18:
Hello,

As part of the IEP v2.6.0 release, we have added the `ESP-IDF: Python Clean` action on the project(right-click on the project to see this option),  this could be useful to resolve the python interpreter issues when switching IDF versions.

More on this.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-py.html#clean-the-python-byte-code-python-clean

[image]

closed 2022-08-18 by kolipakakondal