espressif/idf-eclipse-plugin #504: Tools Manager Install Tools fails because it doesn't have permissions to install udev rules in /etc/udev/

gtjoseph, 2022-03-07:
**Describe the bug**
```
Copying OpenOCD Rules
Copying File: /home/gtjoseph/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/contrib/60-openocd.rules to destination: /etc/udev/rules.d/60-openocd.rules
An internal error occurred during: "Installing tools...".
Invalid thread access
```
**To Reproduce**
Steps to reproduce the behavior:
1. Go to Espressif, ESP-IDF Tools Manager.
2. Click on 'Install tools.'
3. See error

**Expected behavior**
The tools should install without errors.

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)
[config.txt](https://github.com/espressif/idf-eclipse-plugin/files/8194253/config.txt)

**ESP-IDF Tools Information:**
Help > Espressif Tools Manager > List Installed Tools
```
Executing /home/gtjoseph/.espressif/python_env/idf4.4_py3.10_env/bin/python /usr/src/mcu/esp32/esp-idf/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)
```

**IDF Eclipse Plugin Version:** 
Eclipse > About Eclipse > Installation Details > Installed Software > Search for "espressif"
[image]

**Eclipse Error log:** 
Window > Show View > Error Log

[error_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/8194272/error_log.txt)

kolipakakondal, 2022-03-09:
Hi @gtjoseph Thanks for reporting an issue. While we fix this, could you manually copy a file from `/home/gtjoseph/.espressif/tools/openocd-esp32/v0.11.0-esp32-20211220/openocd-esp32/bin/../share/openocd/contrib/60-openocd.rules` to `/etc/udev/rules.d/60-openocd.rules` to avoid OpenOCD debugging issues

gtjoseph, 2022-03-09:
Already done, thanks!

mthii, 2022-03-15:
Hi, I have same issue. As for the rule file, I registered it manually. But the idf eclipse installer failed the required and remaining files to be installed ? Idf build always fails with this message 'Build not configured correctly'. I'm looking for the installer script to be fix. Regards

alirana01, 2022-09-12:
Fixed in latest release

closed 2022-09-12 by alirana01