espressif/idf-eclipse-plugin #506: Errors exist in the required projects

franz-ms-muc, 2022-03-09:
When try flash, 
there appears: 
[image]

franz-ms-muc, 2022-03-09:
this appears sometimes also after a 0 error, 0 warning build. 

why ?

kolipakakondal, 2022-03-09:
Hi @franz-ms-muc It seems to be the project is having errors, could you check the **Problems** view?

franz-ms-muc, 2022-03-09:
YES. 

in the Problems view i see 18 "semantic" Errors. 

but the Project build with 0 warnings /0 Errors ...

kolipakakondal, 2022-03-09:
Ok, are they valid errors (semantic errors) which could be fixed or wrongly reported by the indexer? If it's wrongly reported by the indexer, we could investigate further.

Could you provide a sample piece code where it's showing "semantic" Errors? One such case was reported by [here](https://github.com/espressif/idf-eclipse-plugin/issues/462)

gtjoseph, 2022-03-16:
I have the same issue.  I've had to completely turn off Code Analysis to stop the message.  I'm not sure it's the indexer itself though.  For instance, if the UI shows an unresolved symbol error for a function and I ctrl-click on the function call, I do get taken right to the function declaration.   Re-running the indexer usually does NOT change the errors reported but sometimes doing an explicit validation on a file does.  Sometimes, closing and reopening the file or project also fixes things temporarily.  The issues always come back though.

I'll get some logs and screenshots.

kolipakakondal, 2022-03-17:
Hi @gtjoseph Thanks for reporting. Could you share with us the sample template where it's reporting unresolved symbol issues to look further on this?

gtjoseph, 2022-03-17:
I'll try and get something together today.

gtjoseph, 2022-03-18:
I started from scratch...
* New install of Eclipse 2021-09
* New install of idf-eclipse-plugin
* New workspace
* Create new Espressif IDF Project
* Selected template storage/sd_card/sdmmc
* Created target for esp32s3
* Opened main/sd_card_example_main.c for the first time...

[image]

A build hadn't been done yet so the missing includes are expected.

* Started a build which completed successfully.

[image]

The includes have all been found but there are still unresolved symbols.

* Start a flash

[image]

* Pressed "Proceed" and the flash completed successfully.

* Re-run the indexer

[image]

Everything except this one define is resolved.

* CTRL-click on SDMMC_SLOT_FLAG_INTERNAL_PULLUP

[image]

I get taken directly to the definition in sdmmc_host.h.

* Closed the project
* Did an idf fullclean from a shell
* Reopened the project and sd_card_example_main.c

[image]
[image]

Now there's a different set of errors.

This is only a small sample project.  In my larger projects this is way more intrusive.

There are no errors in the Eclipse log BTW.

kolipakakondal, 2022-03-21:
Hi @gtjoseph Thanks for the sharing. Let us look into the errors.

f-hoepfinger-hr-agrartechnik, 2022-03-21:
today for Example: 

esp_pthread_get_default_config();

underlined red. 

comment: 
Problem description: Invalid arguments ' Candidates are: {C:\Users\Franz\esp-
 idf-v4.4\components\pthread\include\esp_pthread.h:885} 
 esp_pthread_get_default_config() '

Henry586, 2023-04-23:
After re-run the indexer by click menu Project->C/C++ Indexer->Freshen All Files,
the 'Semantic Error' disappeared.
And project properties: C/C++ General->Indexer->Enable indexer is ticked.