espressif/idf-eclipse-plugin #509: Switching to different Serial Port in the ESP Target selector does not take effect

tsctrl, 2022-03-13:
**Describe the bug**
I have two ESPProg connected to flash my device, once i successfully flash device using first ESP Prog i was unable to flash the same project to my second ESP Prog

**To Reproduce**
Steps to reproduce the behavior:
1. Flash to first ESP Prog by selecting the Serial Port in the ESP Target selector
2. Once completed, select Serial port for 2nd ESP Prog the ESP Target selector and start flashing
3. Result, the IDE was flashing firmware to the 1st ESP Prog

**Expected behavior**
Should be able to flash to the target com port specified in the ESP Target

**Screenshots**
[image]

**Environment Information:** 
Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.16.4 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.10.0-esp32-20211111 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.1 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 3.7 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
ESP-IDF Eclipse Plugin
Version: 2.4.2.202203081345

kolipakakondal, 2022-03-15:
Hi @tsctrl Thanks for reporting this. I could reproduce the issue, let us check and get back to you with more details.

sigmaaa, 2022-03-15:
Hi @tsctrl,

while we fixing it, can you try to edit the launch configuration and change the port in `Arguments:` line, please? 
[image]

tsctrl, 2022-03-15:
yup sure

closed 2022-06-13 by sigmaaa