espressif/idf-eclipse-plugin #511: Problems with Install ESP-IDF Components

enrydj, 2022-03-18:
Eclipse IDE for C/C++ Developers - 2021-12 on Linux
esp-idf eclipse plugin 4.4 (looks correctly installed, example projects run correctly)

Problem is when installing components with Install ESP-IDF Components: managed_components directory is created and files are present, but build does not include them.

Steps to reproduce the problem:
create a new project from i2s_basic template
Install ESP-IDF Component: install BUTTON
in file i2s_example_main.c try to include iot_button.h file

File is not found while building (how to write? just iot_button.h or relative path?)

It can be found by correcting CMakeLists.txt file in main directory:
idf_component_register(SRCS "i2s_example_main.c" "../managed_components/espressif__button/button_adc.c" "../managed_components/espressif__button/button_gpio.c" "../managed_components/espressif__button/iot_button.c"
                    INCLUDE_DIRS "." "../managed_components/espressif__button/include")

In that case build fails because configuration defines are not found:
../managed_components/espressif__button/button_adc.c:38:32: error: 'CONFIG_ADC_BUTTON_MAX_BUTTON_PER_CHANNEL' undeclared here (not in a function); did you mean 'ADC_BUTTON_MAX_CHANNEL'?
and other errors.

Did I miss anything?

enrydj, 2022-03-22:
@alirana01 Please could you explain me how and where I have to run the command? Thank you.

alirana01, 2022-03-23:
@enrydj depending upon the os you are using there are two scripts that you need to run from the command line in the esp-idf directory 
first run the install script based on your os and env
- install.ps1 (windows powershell) install.bat (windows cmd) install.sh (linux)

then run export script just like install based on your os and env.

Next you can go to the project root directory and run the command idf.py reconfigure

Alternatively you can select the ESP-IDF Terminal from the eclipse and simply just switch to the project directory and run the command idf.py reconfigure as the terminal in eclipse already contains the variables and exports needed for the env given that idf tools were installed in that particular eclipse environment.

https://user-images.githubusercontent.com/85216275/159665882-174565ba-b881-471d-9866-43b445a81e88.mp4

enrydj, 2022-03-26:
@alirana01 Thank you, I was able to link iot_button.h file using ESP-IDF Terminal. But before I needed to add idf.py file execution permission. By default it could not be executed.

I tried to create a new project and adding the button component with idf.py execution permission enabled, to verify if that was the problem, but I had another issue:

the ESP-IDF entries of right click menu of the projects are no more present, either in new or existing projects. Do you please know how I can see them?

alirana01, 2022-03-29:
@enrydj I am not able to understand which menu you are referring to. Some screenshots or screen recording will be helpful or we can schedule a call.
ali.azamrana@espressif.com

enrydj, 2022-04-02:
@alirana01 Solved menu problem: I have reset perspective.
Window -> Perspective -> Reset Perspective...
Now all ESP-IDF entries are present.
Now I can work! Thank you!

enrydj, 2022-04-02:
@alirana01 Solved menu problem: I have reset perspective.
Window -> Perspective -> Reset Perspective...
Now all ESP-IDF project menu entries are present.

alirana01, 2022-04-04:
Closing issue will be fixed with future release

closed 2022-04-04 by alirana01