espressif/idf-eclipse-plugin #541: Selecting a Example is slow in Linux

franz-ms-muc, 2022-06-02:
When you say: 

"Create a new IDF Project" 

and want to use one of the Examples as a Template, 

this Selection Dialog is Very slow in Linux, and leads somtimes to crash of Eclipse. 

while same is fast in Windows.

franz-ms-muc, 2022-06-02:
another experience: 

if you use the "filter text" it is fast. 

if you navigate the Folders it is slow.

sigmaaa, 2022-06-08:
Hi @franz-ms-muc,

I was not able to reproduce this issue on my virtual box ubuntu. Selection can be slow due to low heap memory value. It's possible to change this value for eclipse in the eclipse.ini file, arguments are -Xms and -Xmx. 
Also, what can help to fix the issue is the `hs_err_pidXXXXX.log` file, which is generated, when eclipse crashed. You can find more details [here](https://wiki.eclipse.org/FAQ_Where_can_I_find_that_elusive_.log_file%3F).
Could you provide us with an error log so we can better understand what could be the cause? Thank you

franz-ms-muc, 2022-06-08:
I will make.

Holen Sie sich Outlook für Android<https://aka.ms/AAb9ysg>

[https://www.meisterschulen-mchn.de/images/meisterschulen/allgemein/4-logos/logofc.jpg]

Meisterschulen am Ostbahnhof
Mühldorfstr. 6
81671 München

Tel.: 089 416002 - 0
Fax: 089 416002 - 111

Internet: www.meisterschulen-münchen.de<http://www.meisterschulen-mchn.de>

diplfranzhoepfinger, 2022-07-27:
i tested on a newly installed Ubuntu 22.04 LTS, and the issue is gone.

closed 2023-09-22 by sigmaaa

diplfranzhoepfinger, 2024-01-17:
Hi, 

the issue is Back on Ubuntu 23.10

diplfranzhoepfinger, 2024-06-26:
logfile here: 

https://gist.github.com/diplfranzhoepfinger/39b97d7a3392195f7cbf0066389e5c22