espressif/idf-eclipse-plugin #551: esp32c3 toolchain does not show up in configuration window

pn21, 2022-06-21:
I am trying to create a new project on my work laptop which uses esp32c3-mini. I have installed everything as shown in the below pictures (starting from “build environment”) however as shown (highlighted in red filename: "Configuration.jpg") below I do not get the option of selecting the esp32c3 toolchain. I have also “added” the target as esp32c3 (did not show by default, only showed esp32 and esp32s2. Since the below toolchain setting is wrong my code does not compile correctly. What do I need to do to get it working?
I am using esp-idf 4.3.1

Thanks
Pankaj

[image]
[image]
[image]
[image]

kolipakakondal, 2022-06-22:
Hi @pn21 

Could you verify the launch target you've created for `esp32c3` in the `New ESP Target` dialog?
 
[image]

[image]

kolipakakondal, 2022-08-18:
Hello, Closing this issue as there is no response. Feel free to report us back if you still have a problem. Thanks

closed 2022-08-18 by kolipakakondal