espressif/idf-eclipse-plugin #557: ESP-IDF tool installation fails in Eclipse

klausbng, 2022-06-25:
Tried a fresh installation of ESP-IDF in Eclipse as my previous one including Eclipse version was quite old.
- Installed eclipse-cpp-2022-06-R-win32-x86_64
- Installed esp-idf-v4.4.1
- Install all offered packages in Eclipse:
    - CMake Editor
    - Eclipse SWTChart
    - Embedded C/C++ CMSIS Packs Core Plug-in
    - Embedded C/C++ Core
    - Embedded C/C++ GDB JTAG Debugging
    - Espressif IDF Plugins for Eclipse
    - Nebula XViewer Widget
- Go to Expressif and Configure existing ESP-IDF
- Answer question to add additional tools with "yes"
- Offered paths to ESP-IDF, Git, Python are correct -> Install Tools

Result:
`Installing tools...
This can take a while. Please be patient.
Executing C:\Users\klaus\AppData\Local\Programs\Python\Python37\python.exe D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py install all
Traceback (most recent call last):
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1858, in <module>
    main(sys.argv[1:])
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1854, in main
    action_func(args)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1416, in action_install
    tools_info = load_tools_info()
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 946, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 964, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 814, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings

Executing C:\Users\klaus\AppData\Local\Programs\Python\Python37\python.exe D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py install-python-env
Python 3.7.3
pip 20.1 from C:\Users\klaus\.espressif\python_env\idf4.3_py3.7_env\lib\site-packages\pip (python 3.7)

Traceback (most recent call last):
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1858, in <module>
    main(sys.argv[1:])
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1854, in main
    action_func(args)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1570, in action_install_python_env
    wheels_dir = get_wheels_dir()
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1473, in get_wheels_dir
    tools_info = load_tools_info()
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 946, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 964, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 814, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings

pip.exe executable not found. Unable to run `pip.exe install websocket-client`
Executing C:\Users\klaus\AppData\Local\Programs\Python\Python37\python.exe D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py export --format=key-value
Traceback (most recent call last):
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1858, in <module>
    main(sys.argv[1:])
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1854, in main
    action_func(args)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 1175, in action_export
    tools_info = load_tools_info()
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 946, in load_tools_info
    return parse_tools_info_json(tools_info)  # type: ignore
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 964, in parse_tools_info_json
    tool = IDFTool.from_json(tool_dict)
  File "D:\espressif\esp-idf-v4.4.1\tools\idf_tools.py", line 814, in from_json
    raise RuntimeError('supported_targets for tool %s is not a list of strings' % tool_name)
RuntimeError: supported_targets for tool xtensa-esp32-elf is not a list of strings
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
Configured CMake toolchain. You can check Preferences > C/C++ > CMake
Install tools completed.`

**Environment Information:** 
[eclipse_environment.txt](https://github.com/espressif/idf-eclipse-plugin/files/8984584/eclipse_environment.txt)

**ESP-IDF Tools Information:**
[tools-info.txt](https://github.com/espressif/idf-eclipse-plugin/files/8984586/tools-info.txt)

**IDF Eclipse Plugin Version:** 
  Espressif IDF Plugins for Eclipse	2.5.0.202206080313	com.espressif.idf.feature.feature.group	

**Eclipse Error log:** 
[eclipse.log](https://github.com/espressif/idf-eclipse-plugin/files/8984589/eclipse.log)

kolipakakondal, 2022-06-29:
Hello @klausbng 

It seems to be an issue with the tools files clean up after switching the idf version from 3.3.2 to 4.4.1

Could you delete D:\espressif\esp-idf-v4.4.1 folder and git clone again!

kolipakakondal, 2022-06-29:
Hello,

We could recommend using [espressif-ide-setup-espressif-ide-2.5.0-with-esp-idf-4.4 ](https://dl.espressif.com/dl/esp-idf/?idf=4.4) if you are using multiple versions of esp-idf and planning to use them side by side. This will take care of installing all the prerequisites.