espressif/idf-eclipse-plugin #579: FatFS missing module for directory functions e.g. f_getcwd at link time

ralphde, 2022-07-27:
**Describe the bug**

undefined reference to `f_getcwd'	in sd_card_example_main.c or any other code wanting to use this library	

by just adding
`    

    char myDir[128];
    ret = f_getcwd(myDir, sizeof(myDir));

`

**To Reproduce**
add above code to the sample project

**Expected behavior**
module not missing at link time

Eclipse > About Eclipse > Installation Details > Configuration > Copy to Clipboard (Attach as a file)

**ESP-IDF Tools Information:**
Executing C:\Users\ralph\.espressif\python_env\idf4.4_py3.10_env\Scripts\python.exe C:\_esp-idf-v4.4\tools\idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
Version: 2.4.1.202202151702

ralphde, 2022-08-02:
@kolipakakondal  please look into this

igrr, 2022-08-02:
@ralphde This doesn't seem to be an idf-eclipse-plugin issue. The function f_getcwd is not compiled because fatfs library is configured with `FF_FS_RPATH=0`. In order for this function to be available, FF_FS_RPATH has to be set to 2.
If you need this feature, please consider opening a feature request at https://github.com/espressif/esp-idf/issues, explaining the use case for enabling it.

kolipakakondal, 2022-09-30:
Hello, Closing this issue as this is not specific to idf eclipse plugin. As mentioned by Ivan, if you need this feature please raise a request directly under the [esp-idf](https://github.com/espressif/esp-idf/issues,) repo.

closed 2022-09-30 by kolipakakondal