espressif/idf-eclipse-plugin #587: Version 2.6.0 ESP-IDF Eclipse plugin still has indexer issues

mbratch, 2022-08-21:
I simply cannot get rid of the myriad of GUI C++ errors in many of my project files. The project compiles fine, so the flagged issues are falsely claimed problems. I just upgraded to 2.6.0 of the plugin but the problem still persists.

Here's just one example:

[image]

I've tried doing a fullclean and every command available under the C/C++ Indexer menu item and nothing fixes it. It's been very frustrating because a lot of the great GUI features such as auto-completion or looking up a function or method simply don't work if it thinks there's an error associated with it.

To reproduce, I just have to open my project and observe.

**Expected behavior**
I would expect reindexing at least to clear the errors. This actually used to work many months ago.... that is, I would get the false erros but used to be able to clear them with reindexing. THis hasn't worked in a long time.

**Environment Information:** 
See attached file.
[EclipseConfig.txt](https://github.com/espressif/idf-eclipse-plugin/files/9389862/EclipseConfig.txt)

**ESP-IDF Tools Information:**
The comand "Help > Espressif Tools Manager > List Installed Tools" doesn't exist. I used "Espressif > ESP-IDF Tool Manager > List installed tools":

> Executing C:\Users\mdbra\AppData\Local\Programs\Python\Python39\python.exe C:\Users\mdbra\.espressif\frameworks\esp-idf-v4.4.2\tools\idf_tools.py list
> * xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
>   - esp-2021r2-patch3-8.4.0 (recommended, installed)
> * xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
>   - esp-2021r2-patch3-8.4.0 (recommended, installed)
> * xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
>   - esp-2021r2-patch3-8.4.0 (recommended, installed)
> * xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
>   - 12.0.1-d9341b81fc (recommended, installed)
> * riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
>   - esp-2021r2-patch3-8.4.0 (recommended, installed)
> * esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
>   - 2.28.51-esp-20191205 (recommended, installed)
> * esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
>   - 2.28.51-esp-20191205 (recommended, installed)
> * cmake: CMake build system
>   - 3.23.1 (recommended, installed)
> * openocd-esp32: OpenOCD for ESP32
>   - v0.11.0-esp32-20220411 (recommended, installed)
> * ninja: Ninja build system
>   - 1.10.2 (recommended, installed)
> * idf-exe: IDF wrapper tool for Windows
>   - 1.0.3 (recommended, installed)
> * ccache: Ccache (compiler cache)
>   - 4.3 (recommended, installed)
> * dfu-util: dfu-util (Device Firmware Upgrade Utilities)
>   - 0.9 (recommended, installed)

**IDF Eclipse Plugin Version:** 
2.6.0

**Eclipse Error log:** 
Attached
[EclipseError.log](https://github.com/espressif/idf-eclipse-plugin/files/9389864/EclipseError.log)

mbratch, 2022-09-05:
Well I think I finally got it resolved. There are several places that the ESP-IDF framework path needs to be manually changed in the settings any time the framework is updated to a new version. I finally spotted all of them. I only wish someone had just told me to check for that. Maybe someone else will see this and it will help them.

mbratch, 2022-09-05:
I found a few places in Eclipse settings for the ESP-IDF framework path and updated those as they weren't up to date. They must all be hunted down and changed manually after upgrading the framework version. This seemed to clear up the false visual errors temporarily, but then they inexplicably came back for now evident reason.

mbratch, 2022-09-15:
Well I spoke too soon. There are still inexplicable errors being noted, just not quite as many as before.

mbratch, 2022-10-16:
One thing I discovered is that the indexer doesn't seem to be smart about nested includes. So if I do this in my source file:

`#include <foo.h>`

And inside of `foo.h` I have:

`#include <bar.h>`

And inside `bar.h` there's a symbol that I'm referencing in my original source file, it will be flagged as unresolved by the indexer, but the compile of course works fine.

gregtoth, 2023-07-24:
I've noticed that on every other launch of Espressif-IDE, the indexer works, symbols resolve, and Open Declaration drill-down works as expected. But for the other launches the indexer is broken, there are tons of unresolved symbols listed in Problems, and Open Declaration doesn't work. Example: Launch ide and indexer is broken. Quit IDE, then launch again, and indexer is okay. Quit ide, then launch again, and indexer is broken. This cycle repeats over and over and seems to be 100% repeatable with clean install of Espressif-IDE 2.10.0, IDF v5.1, and blink example project. Maybe a clue as to the underlying indexer issue?