espressif/idf-eclipse-plugin #588: esptool lock the com port

MrHause, 2022-08-23:
Hello,
I cannot deal with locked com port during flashing the esp. It always takes my port COM31 to flash the board but the esp is visible under other port number. I tried deleting esp target, creating new ones, changing port but it always takes the same port. I use eclipse cpp with esp-idf plugin. 

Eclipse version: Version: 2022-03 (4.23.0)

esp-idf plugin:   Espressif IDF Plugins for Eclipse 2.4.2.202203081345

sigmaaa, 2022-08-23:
Hi, @MrHause.

I think this problem was fixed in 2.5.0 release, where we changed the port option in the arguments field for the project configuration. 
To fix your issue on your version, try to edit the project configuration by clicking on that wheel:
[image]
 Then, in the arguments field, you will probably see the locked port, which you have to change:
 
[image]

If you choose to update the plugin to the latest version, you will have the arguments line like this:
D:\work\esp-idf\tools\idf.py -p ${serial_port} flash.

MrHause, 2022-08-23:
Thank you very much for fast response. That fixed my problem, maybe later I will try to update plugin.

kolipakakondal, 2022-08-23:
As the issue is fixed, closing the ticket

closed 2022-08-23 by kolipakakondal