espressif/idf-eclipse-plugin #594: Import project does not work for ADF examples or other existing projects

bs-eng, 2022-09-12:
https://github.com/espressif/idf-eclipse-plugin/issues/new

Import project does not work for ADF examples or other existing projects

Hello!

I have been programming on a project in ESP-ADF for the last few years. So far I was able to take the project with after upgrading to a higher version of ADF/IDF.

Now I updated to version ADFV2.4.1 which includes by default IDFv4.4.2
I order to do so I needed to throw out my complete eclipse and ADF/IDF setup as something went very wrong with the upgrade as usual. So I decided for a clean install. (yes, I removed also all hidden setup/config files of eclipse as well as of espressif SDK).

My install:
Operating System: windows 10
Java Runtime Version: 14.0.2+12-46
Eclipse Version: 4.24.0.v20220607-0700
Eclipse CDT Version: 10.7.0.202204131742
IDF Eclipse Plugin Version: 2.6.0.202208110735
fatal: not a git repository: ../.git/modules/esp-idf
WARNING: Git version unavailable, reading from source
ESP-IDF v4.4.2
Python set for IDF_PYTHON_ENV: Python 3.8.3rc1

(Git is installed and running fine - I dont know why the plugin does not recognize it when listing installed tools - bt this is not the point of this posting.)

First test: **create** new IDF project from an example project, build and run it.
=> I chose hello-world and it compiles and runs just fine.

Second test: **import** an ADF example project
=> I chose the pipeline_bt_sink example
- import gives the first issues already: every subfolder is recognized as a seperate nested project => as this is not the case I choose to import only the top level project
- then setting launch target => does not work because eclipse does not show any launch targets in the drop down list (as was the case with the IDF example project before)
- so I create a new launch target => that does not get listed/recognized (meaning the drop down never shows it)

Third test: **create** and ADF example project
=> I chose again the pipeline_bt_sink example
- copy the example project into a folder of name pipeline_bt_sink
- create new IDF project and choosing as location the folder pipeline_bt_sink
- launch target is populated
- build works
- flash/run works fine
- when checking what the software does on the board, I notice that no subfolders have been compiled/included in the project
- upon second build there a plenty of errors about not found symbols and missing include paths

Question
Is this a bug in the espressif plugin or are ADF projects somehow not supported?
What would be the (new) recommended way to import existing code as a project?

Thanks!
JR

kolipakakondal, 2022-09-22:
I've added this to the documentation so that others could also benefit, please find it [here](https://github.com/espressif/idf-eclipse-plugin/blob/3e65185b46bd9c27cab5121ef27c5628d283e0c0/FAQ.md#how-to-configure-esp-adf-in-espressif-ide)

bs-eng, 2022-09-22:
Hi Kondal!
Thanks for looking into it!

Indeed it solved the issue of importing. For me it was new that the standard Eclipse Import function now works - before there used to be an explicit menu item "Import existing Espressif ESP32 project"

The only thing that remains is that menu item "Espressif > ESP-IDF Tools Manager > List installed tools" claims that not all tools are installed. Just thought I mention it, in case it is not meant to behave like that. For me it makes no difference as long as the toolchain works.

Thanks a bunch!
Cheers
JR

kolipakakondal, 2022-09-22:
Good to know that it's working!

> xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
> esp-2021r2-patch3-8.4.0 (recommended, installed)
> xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
> esp-2021r2-patch3-8.4.0 (recommended, installed)
> xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
> esp-2021r2-patch3-8.4.0 (recommended, installed)
> xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
> 12.0.1-d9341b81fc (recommended, installed)
> riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
> esp-2021r2-patch3-8.4.0 (recommended, installed)
> esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
> 2.28.51-esp-20191205 (recommended, installed)
> esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
> 2.28.51-esp-20191205 (recommended, installed)
> cmake: CMake build system
> 3.23.1 (recommended, installed)
> openocd-esp32: OpenOCD for ESP32
> v0.11.0-esp32-20220706 (recommended, installed)
> ninja: Ninja build system
> 1.10.2 (recommended, installed)
> idf-exe: IDF wrapper tool for Windows
> 1.0.3 (recommended, installed)
> ccache: Ccache (compiler cache)
> 4.3 (recommended, installed)
> dfu-util: dfu-util (Device Firmware Upgrade Utilities)
> 0.9 (recommended, installed)

Everything shows it's **installed**. Did I miss anything here?

> Additionally I noticed that automatically all IDF components are copied into the project as empty folders upon build. So there appears a new folder "esp_idf_components" in the project which contains all idf components folders (without the actual code and include files).

I have added some info [here](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#why-esp_idf_components-folder-is-created-in-the-project) on why esp_idf_components are added to the project. Please check.

> (Git is installed and running fine - I dont know why the plugin does not recognize it when listing installed tools - bt this is not the point of this posting.)

I will take a note of this. From the functional perspective, we could ignore this!

bs-eng, 2022-09-23:
Thank you so much for adding the outcome already in the official docu!

FYI: the message that not everything is installed occurs only when checking from the menu:
[image]

Then the message is - even though IDF-PATH is set:
[image]

The list I posted originally I got from invoking the check manually from the command line.

Cheers
JR

kolipakakondal, 2022-09-30:
Closing this issue as the original issue of importing and building ESP-ADF example is working fine, will track the erroneous tools message prompt issue separately, and the fix will be provided in the next release.

closed 2022-09-30 by kolipakakondal