espressif/idf-eclipse-plugin #599: freertos/xtensa_api.h no such file or directory

HammerShankiz, 2022-09-25:
Eclipse cannot build projects that include freertos/xtensa_api.h, such as a2dp_sink demo or a2dp_source demo projects.

Running Windows 11:
Install Java 19
Install Python 3.10
Install git for windows 2.37.3
Install ESP IDF 4.3.4
Install Eclipse IDE for C/C++ Developers V 2022-06 (4.24.0)
Install IDF Eclipse plugin stable release from  https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
Install toolchain master (I also tried basically every toolchain version from v 4.0 onwards)
open file>new>Espressif IDF project
type in any project name, next
create a project using one of the templates>bluetooth>bluedroid>bt_classic>a2dp_sink
attempt to build project onto esp32c3

The project is expected to compile and flash to my chip  (ESP-32-C3-DevKitM-1), especially since it's a demo template.

Instead, compiled throws:
fatal error: freertos/xtensa_api.h: No such file or directory
10 | #include "freertos/xtensa_api.h"

[Espressif IDE Product Information.txt](https://github.com/espressif/idf-eclipse-plugin/files/9639820/Espressif.IDE.Product.Information.txt)

[a2dpsinkerror.log](https://github.com/espressif/idf-eclipse-plugin/files/9639824/a2dpsinkerror.log)

sigmaaa, 2022-09-26:
Hi @HammerShankiz,
ESP32-C3 has a RISC-V arch and these examples don't support it yet. These projects have an esp32 as a supported target. You can check what targets are supported for a project, by looking into it README.md file, which is available during project creation:
[image]

HammerShankiz, 2022-09-26:
> Hi @HammerShankiz, ESP32-C3 has a RISC-V arch and these examples don't support it yet. These projects have an esp32 as a supported target. You can check what targets are supported for a project, by looking into it README.md file, which is available during project creation: [image]

ope, my bad.

closed 2022-09-26 by HammerShankiz

HammerShankiz, 2022-09-27:
yeah, the chip is marketed as supporting bluetooth 5, which is confusing. I just bought the wrong hardware