espressif/idf-eclipse-plugin #603: mqtt/ssl Sample broken

diplfranzhoepfinger, 2022-10-04:
**Describe the bug**

use the Sample 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/mqtt/ssl
and try to use the Function "New Espressif IDF Project" 
it does not work.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'New Espressif IDF Project'
2. Click on 'mqtt'
3. Scroll down to 'ssl'
4. build
4. See error

```
CMake Error at /home/franz/esp-idf-v4.4.2/tools/cmake/utilities.cmake:108 (target_sources):
  Cannot specify sources for target "mqtt_ssl.elf" which is not built by this
  project.
Call Stack (most recent call first):
  CMakeLists.txt:12 (target_add_binary_data)
```

**Expected behavior**
no error

**Screenshots**
[image]

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

diplfranzhoepfinger, 2022-10-04:
```
Executing /usr/bin/python3 /home/franz/esp-idf-v4.4.2/tools/idf_tools.py list
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system (optional)
  - 3.23.1 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20220411 (recommended, installed)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended, installed)

CDT Build environment variables
PATH: /home/franz/.espressif/tools/xtensa-esp32-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32-elf/bin:/home/franz/.espressif/tools/xtensa-esp32s2-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s2-elf/bin:/home/franz/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin:/home/franz/.espressif/tools/xtensa-clang/12.0.1-d9341b81fc/xtensa-esp32-elf-clang/bin:/home/franz/.espressif/tools/riscv32-esp-elf/esp-2021r2-patch3-8.4.0/riscv32-esp-elf/bin:/home/franz/.espressif/tools/esp32ulp-elf/2.28.51-esp-20191205/esp32ulp-elf-binutils/bin:/home/franz/.espressif/tools/esp32s2ulp-elf/2.28.51-esp-20191205/esp32s2ulp-elf-binutils/bin:/home/franz/.espressif/tools/cmake/3.23.1/bin:/home/franz/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/bin:/home/franz/.espressif/tools/ninja/1.10.2/:/home/franz/.espressif/python_env/idf4.4_py3.8_env/bin:/home/franz/esp-idf-v4.4.2/tools:/home/franz/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
XAUTHORITY: /run/user/1000/gdm/Xauthority
INVOCATION_ID: 325a508fa55f4e77814d42c2b156c2fb
XMODIFIERS: @im=ibus
XDG_DATA_DIRS: /usr/share/ubuntu:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop
GDMSESSION: ubuntu
GTK_IM_MODULE: ibus
DBUS_SESSION_BUS_ADDRESS: unix:path=/run/user/1000/bus
XDG_CURRENT_DESKTOP: ubuntu:GNOME
OPENOCD_SCRIPTS: /home/franz/.espressif/tools/openocd-esp32/v0.11.0-esp32-20220411/openocd-esp32/share/openocd/scripts
JOURNAL_STREAM: 8:75727
SSH_AGENT_PID: 3890
SESSION_MANAGER: local/franz-exone-WORKSTATION-4204HE:@/tmp/.ICE-unix/3926,unix/franz-exone-WORKSTATION-4204HE:/tmp/.ICE-unix/3926
USERNAME: franz
LOGNAME: franz
MANAGERPID: 3585
PWD: /home/franz
IM_CONFIG_PHASE: 1
GJS_DEBUG_TOPICS: JS ERROR;JS LOG
SHELL: /bin/bash
GIO_LAUNCHED_DESKTOP_FILE: /home/franz/.local/share/applications/epp.package.cpp.desktop
GNOME_DESKTOP_SESSION_ID: this-is-deprecated
IDF_COMPONENT_MANAGER: 1
GTK_MODULES: gail:atk-bridge
XDG_SESSION_DESKTOP: ubuntu
SHLVL: 0
OXYGEN_DISABLE_INNER_SHADOWS_HACK: 1
QT_IM_MODULE: ibus
XDG_CONFIG_DIRS: /etc/xdg/xdg-ubuntu:/etc/xdg
LANG: de_DE.UTF-8
XDG_SESSION_TYPE: x11
DISPLAY: :0
XDG_SESSION_CLASS: user
_: /usr/bin/dbus-update-activation-environment
LIBOVERLAY_SCROLLBAR: 0
IDF_PATH: /home/franz/esp-idf-v4.4.2
GDK_CORE_DEVICE_EVENTS: 1
DESKTOP_SESSION: ubuntu
GPG_AGENT_INFO: /run/user/1000/gnupg/S.gpg-agent:0:1
USER: franz
XDG_MENU_PREFIX: gnome-
GIO_LAUNCHED_DESKTOP_FILE_PID: 273526
QT_ACCESSIBILITY: 1
WINDOWPATH: 2
GJS_DEBUG_OUTPUT: stderr
SSH_AUTH_SOCK: /run/user/1000/keyring/ssh
GNOME_SHELL_SESSION_MODE: ubuntu
XDG_RUNTIME_DIR: /run/user/1000
IDF_PYTHON_ENV_PATH: /home/franz/.espressif/python_env/idf4.4_py3.8_env
HOME: /home/franz

Operating System: linux
Java Runtime Version: 16.0.2+7-67
Eclipse Version: 4.20.0.v20210611-1600
Eclipse CDT Version: 10.3.0.202104042017
IDF Eclipse Plugin Version: 2.6.0.202208110735
ESP-IDF v4.4.2-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.10
```

diplfranzhoepfinger, 2022-10-04:
Workaround: 

replacing the Content of CMakeLists.txt with [https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl/CMakeLists.txt](https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/ssl/CMakeLists.txt) and renaming the Project to mqtt\_ssl to get the binary named mqtt\_ssl.elf

diplfranzhoepfinger, 2022-10-04:
@david-cermak

sigmaaa, 2023-09-22:
Hi @diplfranzhoepfinger,
thank you for reporting this. It's fixed now. Verified on 2.11, but should be ok with previous release as well

closed 2023-09-22 by sigmaaa