espressif/idf-eclipse-plugin #604: Espressif-IDE, Linux Install tools fail due to permissions in udev rules in /etc/udev/

AristarchosOfSamos, 2022-10-04:
Hi,
Working with Espressif-IDE on Ubuntu Linux by using Espressif-IDE v2.4.2 download
[https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/Espressif-IDE.md](https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/Espressif-IDE.md)

Got the same problem though as in previous issue #504

Although previously is reported as fixed, the aforementioned Espressif-IDE v2.4.2 download is still producing the same error.
Could someone please do a new/fresh build of the Espressif-IDE download fix the problem?

Regards,

alirana01, 2022-10-04:
@AristarchosOfSamos  thank you for reporting the issue. This occurs basically because the directory /etc/udev/ requires elevated permissions. Can you please try running the IDE with sudo and see if the issue persists? You can also try to manually copy the rules file with terminal using sudo to the /etc/udev directory. The file to copy is **60-openocd.rules** it can be found in the directory where you installed tools inside openocd directory at the path **...../share/openocd/contrib/60-openocd.rules**

Also please check in the IDE preferences C/C++ -> Build -> Environment that you are able to see the IDF_PATH and other variables. If you can attach the screenshots and share the steps to reproduce it will be helpful to diagnose further if this doesn't help.

AristarchosOfSamos, 2022-10-04:
Hi @alirana01, thanks for answering.

I had sudo running the Espressif-IDE and of course it did completed the intall tools stage. Though the problems came aftward since from then on I could not run Espressif-IDE as a plain user, the previous sudo action changed permissions to other eclipse needed files, thus this is not a solution.
I'm working for quite some time now by having manually applied the requred copy files since the failed instruction was indicated in the log.
Also, the environment variables were successfully generated and I am able to build/execute/monitor and debug all examples.

The problem is the uncertainties when this error happens during tools install if something else is still pending and ommitted.

Regards,

alirana01, 2022-10-05:
Hi @AristarchosOfSamos thanks for the response let me get back to you on this we may have to alter the flow somehow to make it more intuitive for the user in terms of reported error messages. Your feedback will help us get a good idea for this. I will keep this issue opened and will ask you for the feedback again after the changes and internal discussions. 

Thanks again for your feedback.