espressif/idf-eclipse-plugin #605: After I clicked Chinese language, it cant boot again.

Yobby0402, 2022-10-07:
**Describe the bug**
I want to change the language to Chinese, after I clicked the button, the software just break and I cant boot again.

[1665139218683.log](https://github.com/espressif/idf-eclipse-plugin/files/9733228/1665139218683.log)

alirana01, 2022-10-07:
Hi @Yobby0402 thank your for reporting the issue.
Can you please use the template to give some more information on the issue

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:** (Not possible in your case as you are not able to start the eclipse but try to see if you can reinstall and share the information from the english version)
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:**  (Not possible in your case as you are not able to start the eclipse)
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 

To further assist can you please share your eclipse.ini file as well and config.ini file

kolipakakondal, 2022-10-07:
Hi @Yobby0402 

> java.version=1.8.0_333

Looks like you're using Java 1.8 which is not supported by the plugin. As mentioned in the [prerequisites](https://github.com/espressif/idf-eclipse-plugin#installing-prerequisites), idf eclipse plugin need Java 11 and higher to work with it.

kolipakakondal, 2022-10-12:
Closing this issue as this caused by the Java version as mentioned above

closed 2022-10-12 by kolipakakondal