espressif/idf-eclipse-plugin #612: Install tools error: "supported_targets for tool xtensa-esp32-elf is not a list of strings"

rbruneau-altyor, 2022-10-19:
After updating my ESP-IDF version, I tried reinstalling the tools on Eclipse ESP-IDF plugin. When reinstalling ESP-IDF in `Espressif > Download and Configure ESP-IDF` menu, I chose my new ESP-IDF directory and the corresponding Python Env executable:
[image]

However, when running install tools, I get the following output:
[image]

What is happening here? What did I do wrong?

Thanks for you help

sigmaaa, 2022-10-27:
Hi @rbruneau-altyor

Sorry for the late reply. Do I understand correctly, that you were updating ESP-IDF by downloading it from this page:
[image]

Or you were using a git command like this:
```
cd $IDF_PATH
git fetch
git checkout vX.Y.Z
git submodule update --init --recursive
```
and then provided updated ESP-IDF by clicking on `Use an existing ESP-IDF version to download, or use an existing ESP-IDF copy`?

If it was by using the git command, please specify which version you updated from. It would help us to reproduce this issue.
Also, if you were using the git command, can you try to use the first approach with downloading esp-idf.
So far I can't reproduce this issue, but you can try this [solution](https://github.com/espressif/idf-eclipse-plugin/issues/557#issuecomment-1169824922). Thank you

rbruneau-altyor, 2022-10-27:
Same problem when downloading ESP-IDF through Eclipse, I have exactly the same error message:

[image]

sigmaaa, 2022-11-01:
Hi @rbruneau-altyor,

I tried to reproduce your problem again but unfortunately, I couldn't. Most probably you need to clean up your environment properly because I see that you running python 4.4 from the .espressif folder( which could be also corrupt due to installation issue) and pip from python 4.2.

The quick thing, that you can try is to delete the .espressif folder and install tools from Eclipse again. (Use python v3.6 or above installed from here https://www.python.org/downloads/). 

Or you can try this: There are multiple options to fix the environment:

- use Online Installer and pick installed ESP-IDF
- use Espressif IDE installer which contains specific ESP-IDF
- use Offline installer with 4.2
https://github.com/espressif/idf-installer#table-with-downloads
The installer should be able to repair the problem or do a clean deployment.

rbruneau-altyor, 2022-11-16:
Sorry for the late reply.

As you suggested, I cleaned the entire toolchain (uninstall Eclipse + deleted esp-idf and .espressif folders), to proceed to a brand new installation with esp-idf v4.3.3. I used Anaconda as my Python interpreter, and it seems like it is working better know. I still don't know why the installation was so messy on my computer.

Thank you for your help