espressif/idf-eclipse-plugin #617: Need separate Debug & Release build output directories at the very least

JBHunterI, 2022-11-02:
The ESP-IDF Eclipse Plugin stores what it builds into a single 'build' directory. I need to be able to build separate 'Debug' & 'Release' builds. For this, the ESP-IDF CLI allows to run idf.py with a '-B <dir>' option but I couldn't get the plugin to do that in Eclipse until after reading this closed issue: 
>
> "Failed to create temporary file" during compile/flash process. Windows path length. (IEP-312) #168
> from [alezancomelit] opened this issue on Dec 18, 2020
>
Hope it was closed because a fix is available since as @alezancomelit said, even with his wonderful "project [Launch] configuration -> Build Settings -> Additional CMake arguments" workaround that avoids a 'build' directory from being created after a successful build, the thing can't be programmed into the target without getting an OpenOCD error:
> 
> Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> Info : only one transport option; autoselect 'jtag'
> Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
> Info : esp_usb_jtag: capabilities descriptor set to 0x2000
> Warn : Transport "jtag" was already selected
> *** dir=C:/...[projectDir]/build file=flasher_args.json args=verify reset exit ***
> C:\...[espIdfDir]\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share/openocd/scripts/target/esp_common.cfg:209: Error: C:/...[projectDir]/**build**/flasher_args.json: No such file or directory
> in procedure 'program_esp_bins' 
> at file "C:\...[espIdfDir]\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share/openocd/scripts/target/esp_common.cfg", line 209
>
So, we can see the output directory path ends in **'build'** instead of the **'Debug'** I had it build into. I just can't find where procedure 'program_esp_bins' is being called. Is there a patch for the ESP-IDF Eclipse Plugin release 2.60, assuming it's the one setting the output directory incorrectly?  Also, if specifying a relative path is not possible, it would be great if we could use something like the ${project_loc} variable after the '-B' to specify an absolute path.

FYI, the OpenOCD page https://openocd.org/doc/doxygen/html/tasks.html#thelisttargets doesn't even list ESP32 as being supported, and the files list after https://openocd.org/doc/doxygen/html/esp32_8c.html doesn't seem to show support for the ESP32-C3.

sigmaaa, 2022-11-02:
Hi @JBHunterI,

Thanks for reporting this, we will fix it asap. For now, you can use this workaround:
edit debug configuration -> in Config options add `-c program_esp_bins <path-to-build-dir> flasher_args.json verify reset` command, like here:
[image]

Just in case, turn off `Flash every time with application binaries`, because it will duplicate this command with the `build` directory in the path.

JBHunterI, 2022-11-02:
Thank you for the prompt & accurate response, @sigmaaa.  I had to modify my Run Launch Mode's Main tab OpenOCD Arguments to:
   `-c "program_esp_bins C:...[projectDir]/Debug flasher_args.json verify reset exit"`
to use **forward-slashes** instead of double back-slashes, and add the **'exit'** to get it to run. Since ${openocd_path} is used earlier in the arguments, I was also hoping to replace **C:...[projectDir]** with **${project_loc:/[projectName]}** but I guess that doesn't work because of the mandatory double-quotes after the -c switch. Finally was able to get it to run that way, though.

For the Debug Launch Mode, I followed your suggestion for the Debugger and Startup tabs, but I also had to change its C/C++ Application setting from **build\\[projectName]** to **${resource_loc:/[projectName]/Debug/[projectName].elf}**. Then, I was able to debug my code running on my target. I left the Startup tab's Enable ARM semihosting checked, although I'm not sure whether it should. Can you clarify this?

Also, now that I have Run & Debug Launcher Modes, I duplicated the latter to create one that I can use to only connect to a running target without restarting it. I followed the Startup tab's pop-up comments that appear when I hover my mouse cursor over its **Initial Reset** and **Pre-run/Restart reset** checkboxes, saying those 2 boxes should not be checked. I also disabled the Main tab's **Disable auto build**. However, the Console tab's "connect [projectName] [ESP-IDF GDB OpenOCD Debugging]" displays:

`
Open On-Chip Debugger  v0.11.0-esp32-20220706 (2022-07-06-15:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:...[projectPath]/Debug/bootloader/bootloader.bin at 0x0
**Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED**
Info : esp_usb_jtag: serial (68:B6:B3:A3:CB:B0)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101104
Info : starting gdb server for esp32c3 on 3333
Info : Listening on port 3333 for gdb connections
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
**Info : Reset cause (3) - (Software core reset)**
Info : [esp32c3] Found 8 triggers
Info : Flash mapping 0: 0x10020 -> 0x3c020020, 32 KB
Info : Flash mapping 1: 0x20020 -> 0x42000020, 86 KB
Info : Auto-detected flash bank 'esp32c3.flash' size 4096 KB
Info : Using flash bank 'esp32c3.flash' size 4096 KB
** Programming Started **
Info : PROF: Erased 20480 bytes in 555.993 ms
Info : PROF: Data transferred in 361.504 ms @ 55.3244 KB/s
Info : PROF: Wrote 20480 bytes in 885.007 ms (data transfer time included)
** Programming Finished in 2634 ms **
** Verify Started **
Info : PROF: Flash verified in 374.906 ms 
** Verify OK **
** Flashing done for bootloader/bootloader.bin in 3014 ms **
Flashing C:...[projectPath]/Debug/[projectName].bin at 0x10000
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
** Programming Started **
Info : PROF: Erased 176128 bytes in 1758.02 ms
Info : PROF: Data transferred in 2781.15 ms @ 61.8449 KB/s
Info : PROF: Wrote 176128 bytes in 3409.9 ms (data transfer time included)
** Programming Finished in 5653 ms **
** Verify Started **
Info : PROF: Flash verified in 423.1 ms 
** Verify OK **
** Flashing done for [projectName].bin in 6084 ms **
Flashing C:...[projectPath]/Debug/partition_table/partition-table.bin at 0x8000
Info : JTAG tap: esp32c3.cpu tap/device found: 0x00005c25 (mfg: 0x612 (Espressif Systems), part: 0x0005, ver: 0x0)
Info : Reset cause (3) - (Software core reset)
** Programming Started **
Info : PROF: Erased 4096 bytes in 366 ms
Info : PROF: Data transferred in 121.298 ms @ 32.9766 KB/s
Info : PROF: Wrote 4096 bytes in 665.755 ms (data transfer time included)
** Programming Finished in 1519 ms **
** Verify Started **
Info : PROF: Flash verified in 340.236 ms 
** Verify OK **
** Flashing done for partition_table/partition-table.bin in 1865 ms **
** Total programming time 10967 ms **
** Resetting Target **
[... 113 lines cut ...]
I (114) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=15b98h ( 88984) map
I (131) esp_image: segment 4: paddr=00035bc0 vaddr=40386718 size=04b20h ( 19232) load
I (135) esp_image: segment 5: paddr=0003a6e8 vaddr=50000010 size=00010h (    16) load
I (140) boot: Loaded app from partition at offset 0x10000
I (143) boot: Disabling RNG early entropy source...
I (349) cpu_start: Pro cpu up.
I (357) cpu_start: Pro cpu start user code
I (357) cpu_start: cpu freq: 160000000 Hz
I (357) cpu_start: Application information:
I (360) cpu_start: Project name:     [projectName]
I (366) cpu_start: App version:      1
I (370) cpu_start: Compile time:     Nov  2 2022 09:47:08
I (376) cpu_start: ELF file SHA256:  0a6e1ab74277236f...
I (382) cpu_start: ESP-IDF:          dbcf640-dirty
I (388) heap_init: Initializing. RAM available for dynamic allocation:
[... cut ...]
I (415) spi_flash: detected chip: generic
I (415) spi_flash: flash io: dio
W (415) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (428) cpu_start: Starting scheduler.
`

A breakpoint early in main() confirms my target is being reset. Why is my target still being restarted & re-flashed?  Does it have anything to do with the **Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED** line early on?  How do I achieve this "connect to" Launch Mode?

JBHunterI, 2022-11-04:
Any idea why I get the **Info : Reset cause (3) - (Software core reset)** msg at line 10 followed by ****Programming Started ** Info : PROF: Erased 20480 bytes in 555.993 ms** at line 13 in my **Console tab's "connect [projectName] [ESP-IDF GDB OpenOCD Debugging]"** display above when I only want to connect to my target? Will this still occur with the upcoming updates of ESP-IDF & its Eclipse Plugin?