espressif/idf-eclipse-plugin #621: Feature Request: JTAG flash, requires an additional configuration option

wreyford, 2022-11-04:
Environment: Espressif- IDE 
Version: 2.6.0
Build id: 20220811-0838

When setting up a launch configuration, we are limited to a default set of parameters.

Press edit launch configuration.
Choose the main tab.
Tick the Flash over JTAG checkbox.
then ...
I set the flash voltage to default.
I select the proper target, in my case esp32s3
Then I need to select the board. Only two options currently allowed by default.
(1).  ESP32-S3 chip (via ESP-PROG)
(2). ESP32-S3 chip (via ESP USB BRIDGE)
Lacking is a third option:
(3) ESP-S3 chip (via ESP USB BUILTIN)

I have a custom PCPB built around the ESP32S3 module, and we use the OTG USB 
D- GPIO_NUM_19
D+ - GPIO_NUM_20, 
(+) not connected
GRND connected.         
We can successfully Flash and monitor with this setup, but the standard scripts that are on  offer don't work.
I noticed that there is a script **esp32s3-builtin.cfg** as our setup is not having a bridge.

It took me hours to figure it out.

So I then need to edit the Arguments: to read as follows:
`-s ${openocd_path}/share/openocd/scripts -f board/esp32s3-builtin.cfg`

Please can you amend this, **and also fix for debug configurations.**
This will be highly appreciated, and an oversight as far as I am concerned.

sigmaaa, 2022-11-04:
Hi @wreyford,

Thank you for reporting this. The boards' list is based on the `esp-config.json` file, which is located in `.espressif\tools\openocd-esp32\v0.11.0-esp32-20220706\openocd-esp32\share\openocd\scripts\esp-config.json`. Openocd-esp32 master (and latest release) branch already has this option in the esp-config.json, check [this](https://github.com/espressif/openocd-esp32/blob/master/tcl/esp-config.json). So, to have this option, you need to update your openocd, or add this option to your esp-config.json, or replace this file with the master's version.

To select another version of openocd in the Eclipse (or Espressif-IDE):
Go to preferences -> Global OpenOCD Path -> Select bin folder of new openocd like this:
[image]

after that make sure `OPENOCD_SCRIPTS` variable inside eclipse variables is updated accordingly.
[image]

wreyford, 2022-11-04:
I'm sorry, but that is exaclty the setup that I currently have. My openOCD version the same  and my script path also correct.

[image]

Script path

[image]

 So in my current correct setup, my IDE only offers the following:
(1) ESP32S3  - ESP32-S3 chip (via ESP-PROG)

[image]

(2) ESP32S3 - ESP32-S3 chip (via ESP USB BRIDGE)

[image]

I then have to manually edit the script argument to read as follows to work:

[image]

So please, we need to add another option. I'm also not using the DevKit, but an actual module on a custom PCB as I explained.
I have also got my update settings to update each time I run the IDE.

wreyford, 2022-11-04:
Sorry, now I see what is meant. Those are not the paths for the master you posted, but the current release.
So the IDE is obviously on the latest stable release, and it doesn't have the fix yet.

[image]

When can we expect a release version that will include this master fix. I'm happy to edit for now.

wreyford, 2022-11-04:
I just copied the master config over the current version, and now I have it.
Thx a lot.
[image]

closed 2022-11-04 by wreyford