espressif/idf-eclipse-plugin #629: Low application partition size turn off notification

dizcza, 2022-11-13:
Any way to turn this off? Even when I tick "Do not show more" it still does show each time I debug an app.

https://github.com/espressif/idf-eclipse-plugin/blob/95f5b5e8dcae75e5bb332c368b66335c030fdecc/bundles/com.espressif.idf.core/src/com/espressif/idf/core/build/messages.properties#L18

sigmaaa, 2022-11-14:
Hi @dizcza,

Thanks for reporting this issue. 
Looks like `<workspace>/.metadata/.plugins/org.eclipse.core.runtime/.settings/com.espressif.idf.core.prefs` file is not getting saved after the Eclipse restart. We will fix it soon. 
For now, as a workaround, try extracting this `com.espressif.idf.core.prefs` file to the path mentioned earlier 
[com.espressif.idf.core.zip](https://github.com/espressif/idf-eclipse-plugin/files/10003465/com.espressif.idf.core.zip)

or create this file and put this content in there:
```
DO_NOT_SHOW_MSG=true
eclipse.preferences.version=1
```

sigmaaa, 2022-12-27:
Hi @dizcza,

The fix is available on the master branch. Here is [update](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489501999) from there.

Or Espressif IDE with appropriate build: 
[Espressif IDE for Windows](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502003)
[Espressif IDE for Linux](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502000)
[Espressif IDE for mac](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502002)
[Espressif IDE for mac Arch64](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502001)

dizcza, 2022-12-27:
Feel free to close the issue then.

kolipakakondal, 2023-02-07:
This is fixed with the release IEP v2.8.0 release. Hence closing the issue.

closed 2023-02-07 by kolipakakondal