espressif/idf-eclipse-plugin #632: hello_word example builds with 15 errors in Eclipse's Problem tab, partly outside desired output build directory

JBHunterI, 2022-11-15:
I built your hello_world example from the ESP-IDF Eclipse Plugin v2.7.0 with no changes, & it caused the Problem tab to fill with 15 errors, the comes-up with this pop-up when you try to run it, although it seems to run fine:
[image]

Also, I tried to specify an output directory other than "build", and it still created a "build" directory with only an "ide" directory in it:
------------------------------------ Eclipse 'Console' tab: CDT Global Build Console
Building in: C:\...\eclipseESP_ws\hello_world\build
Configuring in: C:\...\eclipseESP_ws\hello_world\build
cmake -G Ninja -DCMAKE_TOOLCHAIN_FILE=C:\App\ESP\esp-idf-v5.0\tools\cmake\toolchain-esp32c3.cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=ON -DCCACHE_ENABLE=1 -DIDF_TARGET=esp32c3 -B C:\...\eclipseESP_ws\hello_world\Release C:\...\eclipseESP_ws\hello_world
-- Found Git: C:/App/ESP/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/App/ESP/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/App/ESP/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/App/ESP/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
-- Checking Python dependencies...
Constraint file: C:\App\ESP\espidf.constraints.v5.0.txt
Requirement files:
 - C:\App\ESP\esp-idf-v5.0\tools\requirements\requirements.core.txt
Python being checked: C:\App\ESP\python_env\idf5.0_py3.8_env\Scripts\python.exe
Python requirements are satisfied.
-- Project sdkconfig file C:/.../eclipseESP_ws/hello_world/sdkconfig
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: C:/App/ESP/python_env/idf5.0_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "hello_world" version: 1
-- Adding linker script C:/.../eclipseESP_ws/hello_world/Release/esp-idf/esp_system/ld/memory.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_system/ld/esp32c3/sections.ld.in
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.api.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.libgcc.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.newlib.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.version.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/esp_rom/esp32c3/ld/esp32c3.rom.eco3.ld
-- Adding linker script C:/App/ESP/esp-idf-v5.0/components/soc/esp32c3/ld/esp32c3.peripherals.ld
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_netif esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table protobuf-c protocomm pthread riscv sdmmc soc spi_flash spiffs tcp_transport tinyusb ulp unity usb vfs wear_levelling wifi_provisioning wpa_supplicant
-- Component paths: C:/App/ESP/esp-idf-v5.0/components/app_trace C:/App/ESP/esp-idf-v5.0/components/app_update C:/App/ESP/esp-idf-v5.0/components/bootloader C:/App/ESP/esp-idf-v5.0/components/bootloader_support C:/App/ESP/esp-idf-v5.0/components/bt C:/App/ESP/esp-idf-v5.0/components/cmock C:/App/ESP/esp-idf-v5.0/components/console C:/App/ESP/esp-idf-v5.0/components/cxx C:/App/ESP/esp-idf-v5.0/components/driver C:/App/ESP/esp-idf-v5.0/components/efuse C:/App/ESP/esp-idf-v5.0/components/esp-tls C:/App/ESP/esp-idf-v5.0/components/esp_adc C:/App/ESP/esp-idf-v5.0/components/esp_app_format C:/App/ESP/esp-idf-v5.0/components/esp_common C:/App/ESP/esp-idf-v5.0/components/esp_eth C:/App/ESP/esp-idf-v5.0/components/esp_event C:/App/ESP/esp-idf-v5.0/components/esp_gdbstub C:/App/ESP/esp-idf-v5.0/components/esp_hid C:/App/ESP/esp-idf-v5.0/components/esp_http_client C:/App/ESP/esp-idf-v5.0/components/esp_http_server C:/App/ESP/esp-idf-v5.0/components/esp_https_ota C:/App/ESP/esp-idf-v5.0/components/esp_https_server C:/App/ESP/esp-idf-v5.0/components/esp_hw_support C:/App/ESP/esp-idf-v5.0/components/esp_lcd C:/App/ESP/esp-idf-v5.0/components/esp_local_ctrl C:/App/ESP/esp-idf-v5.0/components/esp_netif C:/App/ESP/esp-idf-v5.0/components/esp_phy C:/App/ESP/esp-idf-v5.0/components/esp_pm C:/App/ESP/esp-idf-v5.0/components/esp_psram C:/App/ESP/esp-idf-v5.0/components/esp_ringbuf C:/App/ESP/esp-idf-v5.0/components/esp_rom C:/App/ESP/esp-idf-v5.0/components/esp_system C:/App/ESP/esp-idf-v5.0/components/esp_timer C:/App/ESP/esp-idf-v5.0/components/esp_wifi C:/App/ESP/esp-idf-v5.0/components/espcoredump C:/App/ESP/esp-idf-v5.0/components/esptool_py C:/App/ESP/esp-idf-v5.0/components/fatfs C:/App/ESP/esp-idf-v5.0/components/freertos C:/App/ESP/esp-idf-v5.0/components/hal C:/App/ESP/esp-idf-v5.0/components/heap C:/App/ESP/esp-idf-v5.0/components/http_parser C:/App/ESP/esp-idf-v5.0/components/idf_test C:/App/ESP/esp-idf-v5.0/components/ieee802154 C:/App/ESP/esp-idf-v5.0/components/json C:/App/ESP/esp-idf-v5.0/components/log C:/App/ESP/esp-idf-v5.0/components/lwip C:/.../eclipseESP_ws/hello_world/main C:/App/ESP/esp-idf-v5.0/components/mbedtls C:/App/ESP/esp-idf-v5.0/components/mqtt C:/App/ESP/esp-idf-v5.0/components/newlib C:/App/ESP/esp-idf-v5.0/components/nvs_flash C:/App/ESP/esp-idf-v5.0/components/openthread C:/App/ESP/esp-idf-v5.0/components/partition_table C:/App/ESP/esp-idf-v5.0/components/protobuf-c C:/App/ESP/esp-idf-v5.0/components/protocomm C:/App/ESP/esp-idf-v5.0/components/pthread C:/App/ESP/esp-idf-v5.0/components/riscv C:/App/ESP/esp-idf-v5.0/components/sdmmc C:/App/ESP/esp-idf-v5.0/components/soc C:/App/ESP/esp-idf-v5.0/components/spi_flash C:/App/ESP/esp-idf-v5.0/components/spiffs C:/App/ESP/esp-idf-v5.0/components/tcp_transport C:/App/ESP/esp-idf-v5.0/components/tinyusb C:/App/ESP/esp-idf-v5.0/components/ulp C:/App/ESP/esp-idf-v5.0/components/unity C:/App/ESP/esp-idf-v5.0/components/usb C:/App/ESP/esp-idf-v5.0/components/vfs C:/App/ESP/esp-idf-v5.0/components/wear_levelling C:/App/ESP/esp-idf-v5.0/components/wifi_provisioning C:/App/ESP/esp-idf-v5.0/components/wpa_supplicant
-- Configuring done
-- Generating done
-- Build files have been written to: C:/.../eclipseESP_ws/hello_world/Release
cmake --build . -- -v
[1/808] cmd.exe /C "cd /D C:\...\eclipseESP_ws\hello_world\Release && C:\App\ESP\tools\cmake\3.24.0\bin\cmake.exe -E touch C:/.../eclipseESP_ws/hello_world/Release/project_elf_src_esp32c3.c"
...
[31/808] cmd.exe /C "cd /D C:\...\eclipseESP_ws\hello_world\Release\esp-idf\partition_table && python C:/App/ESP/esp-idf-v5.0/components/partition_table/gen_esp32part.py -q --offset 0x8000 --flash-size 2MB -- C:/App/ESP/esp-idf-v5.0/components/partition_table/partitions_singleapp.csv C:/.../eclipseESP_ws/hello_world/Release/partition_table/partition-table.bin && C:\App\ESP\tools\cmake\3.24.0\bin\cmake.exe -E echo "Partition table binary generated. Contents:" && C:\App\ESP\tools\cmake\3.24.0\bin\cmake.exe -E echo ******************************************************************************* && python C:/App/ESP/esp-idf-v5.0/components/p [... cut ...]
               0 File(s)              0 bytes
               8 Dir(s)  300,690,464,768 bytes free

C:\...\eclipseESP_ws\hello_world> 

I confirmed the exact same project built with idf.py  -B <output_build_dir> from the ESP-IDF CMD shell doesn't create this build\ide\esp-idf-components directory structure in the root of the example project. I'm worried this will interfere with anything I build in the "build" directory, as your build_system -> cmake -> multi_config example will have me do. Any ideas?

prj, 2022-12-14:
Same here.
Installed just the ESP IDF tools, created a project from example template, and nothing gets resolved.
IDE unusable.

I am going to try VSCode, if you look at this issue then it has been unsolved for years now, and getting this plugin to work is some voodoo dance of finding the very special Eclipse version where it works.

Even the espressif-ide installed with their own installer has this problem with Hello World. So this plugin is unusable in reality.

JBHunterI, 2022-12-14:
I've been told VSCode/PlatformIO isn't an option since Espressif seems to have chosen to only officially support Eclipse as the graphical IDE. I'm now trying to use only the ESP-IDF shell command-line tools to build & write to Flash (ex: idf.py build/menuconfig/flash) and debug using openocd + gdbgui/gdbtui. That leaves me free to use my favorite editor/IDE to manage the source files, and only deal with ESP-IDF issues. I'll still have to make sure everyone else in my dev team will be able to do use what I check-in with this IDF Eclipse Plugin, but at at least I won't have to deal with its instability. This workflow is also more portable between development workstation OS's (Windows, Linux, macOS).

prj, 2022-12-14:
I tried VSCode, and it's even worse. If you try to follow what it says, then it doesn't even build. There are no error messages, just lots of things that do absolutely nothing.

I managed to get the Eclipse to work. I think they screwed it up bigtime at some point.
The solution is to use the 2.4.2 version. I am even using v5.0 tools with it.

All I did was remove the 2.7.0 that came with the tools, and drop the 2.4.2 in there instead. Build the project and no errors or problems, everything gets resolved just fine.
Here is a link:
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.2

Get the relevant whole distro under "Espressif IDE" for your OS.
See if the issue is solved for you as well this way.

kolipakakondal, 2022-12-14:
Hi @prj I am sorry that it didn't work out of the box. Would you mind dropping a mail to ide.support@espressif.com so that we could connect over Zoom/Teams(whatever works for you) and understand more about your issue?

kolipakakondal, 2022-12-14:
Hi @JBHunterI 

By default build output is written to `/project-path/build` directory. However, you should be able to change this using the build configuration window through CMake arguments as specified [here](https://github.com/espressif/idf-eclipse-plugin#custom-build-directory)

But there is a catch here, even if you specify a different build directory you still find the `/project/build/ide/esp_idf_components` folder to support editor source code navigation. Please remember actual build-generated artifacts will be written to the custom build directory which you have specified, for example in your case `/project/release`. There shouldn't be a problem with this in resolving headers and indexing. Anyway, will provide a fix for this as soon as possible.

If you still have a problem with unresolved headers, please provide us with a sample example so that we could verify.

Thanks

kolipakakondal, 2022-12-14:
> @kolipakakondal I do not do anything special whatsoever. I simply installed using the Windows online installer, and installed also esp IDE.
> Then started it from c:\Espressif\tools\Espressif-IDE.
> After that I created a project based on SoftAP template. After first build everything is resolved, after a second build nothing is anymore. I did not modify any settings.

I was trying to test this issue on Windows and macOS and I couldn't reproduce this issue. Maybe something else is missing.

**Tested environment:**
- Espressif IDE v2.7.0 ESP-IDF Windows Installer from [here](https://dl.espressif.com/dl/esp-idf/) 
- java 11.0.7 2020-04-14 LTS
- Windows 10

Just in case if it's useful I have uploaded a recording of this complete scenario starting from downloading Espressif IDE v2.7.0 ESP-IDF Windows Installer to building a project.
https://www.youtube.com/watch?v=gU3UHvOyvLE

prj, 2022-12-14:
I can tell you right away what I did differently, I started with the online installer and I used toolset 5.0.
Not the 2nd installer you selected.

Not sure if that affects anything. 2.4.2 is working perfectly for me, and I am using that for now.

If I will have time, I will record the full installation process for you... but I don't really want to break my existing setup.

kolipakakondal, 2022-12-15:
I did further testing with the Online installer

- I have downloaded https://dl.espressif.com/dl/idf-installer/esp-idf-tools-setup-online-2.17.exe which is the latest `Universal Online Installer for Windows`
- Selected ESP-IDF v5.0 and Espressif-IDF during the installation
- Launched Espressif-IDE from the desktop shortcut
- Created `softAp` project from the template list
- Built with `esp32c3` target - build went fine and no errors were reported in the indexing
- Made some changes in the softap_example_main.c and rebuilt the project - build and indexing went fine
- Changed target to `esp32`
- Built again - no errors reported during the build and indexing went fine
- Restarted IDE just to see if that's causing any issues. But I couldn't find any unresolved header issues even after the eclipse restart.

Anyway if you come across the same error please share us the sample project, [error log](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-access-the-error-log), and [parser log](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-access-cdt-parser-error-log).

prj, 2022-12-15:
Did you use the default workspace location or a different, new workspace location?
Also, I use ESP32-S3 target. But I think target did not matter.

kolipakakondal, 2022-12-15:
> Did you use the default workspace location or a different, new workspace location? Also, I use ESP32-S3 target. But I think target did not matter.

I was using the new workspace location and tried with esp32-s3 as well.

kolipakakondal, 2022-12-15:
My bad, there seems to be an issue with the Hello_world template with esp-idf v5.0(not earlier versions). I could find some unresolved symbols. Let us work on this.

Thanks for all the pointers.

Workaround:
Unresolved symbol issues from Hello-world disappear if we move esp_chip_info.h import statement to the top

kolipakakondal, 2022-12-16:
We got more leads on this issue - In IDF v5.0, It was upgraded from -std=gnu++11 to -std=gnu++20, and also upgraded the compiler to the version which supports C++20. Unfortunately, neither Eclipse CDT nor IDF Eclipse plugin has the support for the C++20 language features hence CDT Parser is unable to parse certain files where the new C++20 features are used. That's the reason, you see unresolved symbol issues on certain files with IDF v5.0 but not with the earlier versions of IDF.

For someone If you don't need esp-idf v5.0, you can switch back to v4.4.3 until we get a fix for this.

I will keep you posted!

**p.s:**

This is where the CDT Parser is going wrong
<code>
org.eclipse.cdt.internal.core.parser.ParserException: Encountered an ambiguous node "std::uses_allocator_construction_args<_Tp1>(__a,         std::forward<_Up>(__u))" at /Users/kondal/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/11.2.0/bits/uses_allocator_args.h, line 170 while parsing /Users/kondal/esp/esp-idf/components/nvs_flash/src/nvs_api.cpp     at  
</code>

C++14 and C++20 feature support bugs
https://bugs.eclipse.org/bugs/show_bug.cgi?id=551720
https://bugs.eclipse.org/bugs/show_bug.cgi?id=451086

JBHunterI, 2022-12-16:
I guess this is also the cause for the "Errors running builder 'CDT Core Builder' on project 'hello_world'." error in [IDF Eclipse Plugin does not yet support Eclipse CDT 2022-12 (IEP-826) #654 ](https://github.com/espressif/idf-eclipse-plugin/issues/654) then?

maxbrito500, 2023-03-08:
Same thing here. Had to install version 4.4.3 to remove the syntax error messages from the IDE.

Please add this information on the main page. It is confusing to install the most recent recommended version that is obviously giving these kind of errors.

valkuc, 2023-03-20:
Looks like this is the same as discussed here - https://github.com/espressif/idf-eclipse-plugin/issues/643#issuecomment-1476711686

gregtoth, 2023-07-24:
I've noticed that on every other launch of Espressif-IDE, the indexer works, symbols resolve, and Open Declaration drill-down works as expected. But for the other launches the indexer is broken, there are tons of unresolved symbols listed in Problems, and Open Declaration doesn't work. Example: Launch ide and indexer is broken. Quit IDE, then launch again, and indexer is okay. Quit ide, then launch again, and indexer is broken. This cycle repeats over and over and seems to be 100% repeatable with clean install of Espressif-IDE 2.10.0, IDF v5.1, and blink example project. Maybe a clue as to the underlying indexer issue?