espressif/idf-eclipse-plugin #636: Environment variables defined in Run Configuration was undefined when used in a CMakeList.txt

tsctrl, 2022-11-19:
**Describe the bug**
Environment variables defined in Run Configuration was undefined when used in a CMakeList.txt

[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Run Configuration Dropdown 
2. Click on edit
3. Go to Environment tab
4. Add new env variable TEST with value Test1
5. Run sample cmake below:

```
if (NOT DEFINED $ENV{TEST})
	message(FATAL_ERROR "TEST  IS NOT DEFINED.")
endif()
```

**Behavior**
$ENV variable is undefined

**Expected behavior**
$ENV variable should be captured by CMake during build

**Espressif-IDE Product Information:**
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch3-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.28.51-esp-20191205 (recommended, installed)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
  - 2.28.51-esp-20191205 (recommended, installed)
* cmake: CMake build system
  - 3.20.3 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20211220 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended, installed)

IDF_PATH: \esp-idf\esp-idf-v4.4.1
IDF_PYTHON_ENV_PATH: \.espressif\python_env\idf4.4_py3.8_env
PATH: \.espressif\tools\xtensa-esp32-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32-elf\bin;\.espressif\tools\xtensa-esp32s2-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s2-elf\bin;\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin;\.espressif\tools\xtensa-clang\12.0.1-d9341b81fc\xtensa-esp32-elf-clang\bin;\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch3-8.4.0\riscv32-esp-elf\bin;\.espressif\tools\esp32ulp-elf\2.28.51-esp-20191205\esp32ulp-elf-binutils\bin;\.espressif\tools\esp32s2ulp-elf\2.28.51-esp-20191205\esp32s2ulp-elf-binutils\bin;N\.espressif\tools\cmake\3.20.3\bin;\.espressif\tools\openocd-esp32\v0.11.0-esp32-20211220\openocd-esp32\bin;\.espressif\tools\ninja\1.10.2\;\.espressif\tools\idf-exe\1.0.3\;\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;\.espressif\tools\dfu-util\0.9\dfu-util-0.9-win64;

Operating System: windows 10
Java Runtime Version: 11.0.14.1+10-LTS
Eclipse Version: 4.22.0.v20211124-1800
Eclipse CDT Version: 10.5.0.202109270023
IDF Eclipse Plugin Version: 2.4.2.202203081345
ESP-IDF v4.4.1-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7
Python declared in PATH variable: Python 3.9.0

alirana01, 2022-11-25:
Hi

@tsctrl Thank you for reporting the issue. Can you please try to define the variable in eclipse's settings.
**C/C++ -> Build -> Environment** 
And add the variable there and see if it helps.

Thanks

tsctrl, 2022-11-25:
hi @alirana01 ,

this issue was not resolve. adding variable in c++ env will set global variable for all build and not for specific run build. your suggestion was not solving this issue. adding variable in c++ configuration is similar as adding settings in the cmake files or in machine env variable which is not change for every run configuration build.

i was looking for settings to be dynamically change for each different run build not shared c++ env that was set for all build.

tsctrl, 2022-11-25:
@alirana01 please reopen this issue.

thanks!
@kolipakakondal

igrr, 2022-11-25:
@tsctrl one alternative is to add build-specific settings via CMake arguments in the Additional CMake Arguments field on the Build Settings tab of the Edit Configuration window. If you add an argument such as -DFOO=1, then variable FOO will be defined to 1 in CMake.

[image]

tsctrl, 2022-11-25:
i tried, added the variable in the additional cmake arguments but it will be the same value for all run build, so this approach also does not solve this issue and the same as adding the argument directly on the cmake file.

tsctrl, 2022-11-26:
the objective is to have different env val for different run build.

say, 
run build esp32-foo1 is to build with config variable FOO=1
run build esp32-foo2 is to build with config variable FOO=2

with this we dont have to manually change cmake file for every build. please also give suggestion on other possible approach.

kolipakakondal, 2022-11-28:
> the objective is to have different env val for different run build.
>
> say, run build esp32-foo1 is to build with config variable FOO=1 run build esp32-foo2 is to build with config variable FOO=2 [...]

Hi @tsctrl You can achieve this by duplicating the existing launch configuration. Please follow the below instructions
1. Click on the project launch configuration `Edit` button (Gear icon from the top toolbar)
2. Click on `Duplicate`
3. Change the launch configuration name as you desired
4. Navigate to the `Build Settings` Tab
5. Provide additional arguments in `Additional CMake arguments`
6. Click on Ok

Now you've two launch configurations for your project to address the specific needs.

We will work on the environment variable issue you've mentioned, thanks for bringing this up.

[image]

tsctrl, 2022-11-28:
hi @kolipakakondal i have done as above but the additional cmake arguments will always be the same for all run configuration.

let say i change the cmake additional configuration in run config A. the run config B also will have the same cmake argument. means the same cmake configuration are shared between all run configuration.

this behaviour was invalidates the point of having multuple additional cmake arguments for each run configuration.

igrr, 2022-11-28:
> i have done as above but the additional cmake arguments will always be the same for all run configuration.

I can confirm this! For some reason, modifying "Additional CMake arguments" for one configuration changes this setting for all other configurations...

kolipakakondal, 2022-11-28:
Thanks for verifying @igrr @tsctrl let me fix this and provide a nightly build

kolipakakondal, 2022-12-02:
Hi, I thought I could provide a quick fix to handle this -after spending a couple of hours realized this is a lot more complicated than I thought as the underlying build infrastructure to handle multiple build configurations for a project is missing in Eclipse CDT CMake itself. 

We will take this up on priority to build this feature.  Let me also reopen this bug as the environment variables are not getting parsed during the build.

kolipakakondal, 2023-03-03:
> Hi, I thought I could provide a quick fix to handle this -after spending a couple of hours realized this is a lot more complicated than I thought as the underlying build infrastructure to handle [...]

Issue related to handling multiple build configurations and saving was addressed in IEP v2.8.0 released.

WilliamFrasson, 2024-07-20:
Hi
it seem that the same issue is till present in ESP-IDE 3.0

therre is any update on this issue?