espressif/idf-eclipse-plugin #638: [v.2.7.0] COM port locking after flashing

GTCLive, 2022-12-02:
**Describe the bug**
Hi, I'm using a ESP32-S3 and the JTAG port for both flashing and console monitoring. If I flash while the Terminal is connected it locks the port and the only way to unlock it is to close and restart Eclipse. This was not the case in v.2.6.0 where flashing while having Terminal connected at the same worked quite well. Just started after upgrading to v2.7.0. Only workaround I've found so far is to 1) disconnect from Terminal 2) flash 3) reconnect to Terminal. 

**To Reproduce**
Steps to reproduce the behavior:
1. In Eclipse, connect to COM port via Terminal
2. Compile and flash to board
3. Go back to Terminal and you will see <Connecting...>

**Screenshots**
[image]

[image]

userbogd, 2022-12-10:
I have the same issue after update plugin to version 2.7. Rollback to 2.6 fix this behaviour. As temporary solution i manually disconnect from terminal before flash the device, every time. If forget to do it, it will be neccessary kill python process of eclipse restart.

kolipakakondal, 2022-12-12:
Hello, Sorry for the trouble. Could you please help us to troubleshoot this issue by identifying which process is holding the COM port?

Follow the instructions from here https://knowledge.ni.com/KnowledgeArticleDetails?id=kA03q000000YGw9CAG&l=en-US

userbogd, 2022-12-12:
The instruction you given can't find the process which locked the serial port. But i can point the process, which i killing for unlock. Also i guess you'll be interested in command line, starts this process. `"C:\Users\bogd\AppData\Local\Programs\Python\Python310\python.exe" C:\Users\bogd\esp-idf-v4.4.3\tools\idf_monitor.py -p \\.\COM6 -b 115200 --ws ws://localhost:53635 C:/BOGD-PROJECTS/dispHUB75/build/dispHUB75.elf` 
[image]

kolipakakondal, 2022-12-13:
Hi @userbogd Thanks for sharing that. Most likely this is due to the core dump debugging feature which we introduced in v2.7.0 where websocket process might be holding the port

Will investigate further on this @alirana01

sigmaaa, 2022-12-26:
Hi @userbogd, @troky, @GTCLive,

Looks like this issue is fixed in the PR, which is still in review, but you can try these changes if you want. Here is [update](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489501999). 

Or if you don't want to affect your current Eclipse, you can try these changes on Espressif IDE: 
[Espressif IDE for Windows](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502003)
[Espressif IDE for Linux](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502000)
[Espressif IDE for mac](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502002)
[Espressif IDE for mac Arch64](https://github.com/espressif/idf-eclipse-plugin/suites/10056670355/artifacts/489502001)

After installing Espressif IDE, click on Install Tools first, so environment variables will be set. Thank you

userbogd, 2022-12-26:
for me this fix works fine! thank you so much!

troky, 2022-12-26:
I can confirm it works fine now! Thanks.

GTCLive, 2022-12-27:
@sigmaaa
Apologies for the delay in trying this out and responding (xmas busy). I can also confirm the update works perfectly fine.
Amazing work and support, thank you so much. 

Let me know if you need this case closed now, or if you prefer leaving it open a bit longer for awareness.

sigmaaa, 2022-12-27:
Hi @GTCLive,

thank you for your feedback. Let's keep it open until the next release when we will include these changes. Thank you

ddomnik, 2023-01-02:
> Hi @userbogd, @troky, @GTCLive,
>
> Looks like this issue is fixed in the PR, which is still in review, but you can try these changes if you want. Here is [...]

Thanks for providing a fix. Could you please explain how to use it? 
I am on Windows. I tried to install com.espressif.idf.update.zip via Help > Install new Software, but all configs and the "Espressif" tab went missing after a restart.
Also tried to copy and replace the contents of the espressif-ide-win32.zip.zip into the IDE installation folder. But get JRE/JDK not found when starting.

Thanks

userbogd, 2023-01-03:
> but all configs and the "Espressif" tab went missing after a restart.

If you question is still not resolved. As mentioned in message your quoted, zip archive must be install over previously installed plugin. So, firstly install last released  plugin as link, and after install this patch as zip archived plugin. For me this sequence on Windows works fine.

troky, 2023-01-03:
I am using Eclipse 2022-12 with IDF plugin 2.7.0 and like @userbogd just installed provided zip via Help->Install New Software and everything worked. On Windows.

kolipakakondal, 2023-01-03:
> Eclipse 2022-12 with ID

I would recommend not using Eclipse 2022-12 as this is not officially supported and it also requires Java 17. 

https://github.com/espressif/idf-eclipse-plugin/issues/654

troky, 2023-01-03:
> I would recommend not using Eclipse 2022-12 as this is not officially supported and it also requires Java 17. Instead prefer Espressif-IDE v2.7.0 or Eclipse CDT 2022-09 with IEP plugin update

Yeah, I installed it without knowing about incompatibility but it works fine for me... also installed all possible updates. I guess I just had luck :)

closed 2023-01-05 by kolipakakondal

f-hoepfinger-hr-agrartechnik, 2023-02-08:
> Hi @userbogd, @troky, @GTCLive,
>
> Looks like this issue is fixed in the PR, which is still in review, but you can try these changes if you want. Here is [...]

cool to see a Espressif IDE upcoming for Linux. 

very cool !!!!

f-hoepfinger-hr-agrartechnik, 2023-02-08:
are there Permanent links to the Espressif IDE ?

sigmaaa, 2023-02-09:
> are there Permanent links to the Espressif IDE ?

Hi @f-hoepfinger-hr-agrartechnik 
Espessif IDE different versions are available here https://github.com/espressif/idf-eclipse-plugin/releases

f-hoepfinger-hr-agrartechnik, 2023-02-11:
found it, thanks https://github.com/espressif/idf-eclipse-plugin/blob/master/docs/Espressif-IDE.md#downloads