espressif/idf-eclipse-plugin #643: Eclipse 2022-09 with ESP-IDF Release v5.0 examples compile problem, not resolving all symbols

Hoiss, 2022-12-08:
**Describe the bug**
No compile possible of any Espressif IDF example project due to unresolved symbols from components directory.

**To Reproduce**
Steps to reproduce the behavior:
1. Fresh install of eclipse 2022-09
2. Follow [readme](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md) to install the ESP-IDF and configure Eclipse with the plugins
3. create a new Espressif IDF project, chose hello_world as an example
4. compile the project -> 3 errors:
`Symbol 'CHIP_FEATURE_BLE' could not be resolved	hello_world_main.c	/hello_world/main	line 26	Semantic Error
Symbol 'CHIP_FEATURE_BT' could not be resolved	hello_world_main.c	/hello_world/main	line 25	Semantic Error
Symbol 'CHIP_FEATURE_EMB_FLASH' could not be resolved	hello_world_main.c	/hello_world/main	line 37	Semantic 
`
Please note that these symbols in the hello_world_main.c are marked red as not resolvable. But you can follow them by right clicking and Open Declaration.

**Expected behavior**
Compile without errors, all symbols resolvable

**Screenshot:**
[image]

**Espressif-IDE Product Information:**
[info.txt](https://github.com/espressif/idf-eclipse-plugin/files/10190083/info.txt)

**Eclipse Error log:** 
[eclipse.log](https://github.com/espressif/idf-eclipse-plugin/files/10190223/eclipse.log)

**How to proceed?**
I can't compile any code at the moment.
It seems it misses the ESP-IDF components directory as include, I can find the needed file there. 
But why and how can I solve it? Is this problem related to Eclipse 2022-09?
Looking around I found some guys added e.g. a missed path in the project properties. But I don't have the option there to set this: 
[image]

Hoiss, 2022-12-08:
Concentrating on the problems with the bleprph example I could solve the compile issues by adding
`set(EXTRA_COMPONENT_DIRS /home/trigo/sub/8-ESP32/ESP-IDF/esp-idf-v5.0/examples/bluetooth/nimble/common/nimble_peripheral_utils)` to the CMakeLists.txt file.
But after flashing it, the example outputs a core 0 panic:
`Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled`
Something still goes wrong, but the example is unchanged and should work...

Hoiss, 2022-12-09:
I setup a new Ubuntu 22.10 VM, again ensured the prerequisites are met, installed eclipse (this time 2022-06), ESP-IDF and the plugins via the GUI as stated in the readme.
Then I setup a new project with the bleprph example and compiled:
`CMake Error at /home/trigoVM/esp-idf-v5.0/tools/cmake/project.cmake:243 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /home/trigoVM/eclipse-workspace/common/nimble_peripheral_utils
Call Stack (most recent call first):
  /home/trigoVM/esp-idf-v5.0/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:10 (project)`

I had this already running in the past without problems. The used ESP-IDF was beta at this time and now it's in released state.
Shouldn't these examples work out of the box now?

kolipakakondal, 2022-12-09:
Hi @Hoiss I can reproduce the problem. Instead of creating a project templates list using the new project wizard, import as a project then should work without modifying EXTRA_COMPONENT_DIRS  in the CMakeLists file.

[image]

And Don't select "Copy project into workspace" so that it will correctly refer to the nimble_peripheral_utils dir

kolipakakondal, 2022-12-09:
Or you can also change the EXTRA_COMPONENT_DIRS path in CMakeLists.txt. 

For example:
```
set(EXTRA_COMPONENT_DIRS $ENV{IDF_PATH}/examples/bluetooth/nimble/common/nimble_peripheral_utils)
```

I think this a better approach as you could keep the project anywhere

Hoiss, 2022-12-09:
Thank you @kolipakakondal 
The import is the trick, especially `And Don't select "Copy project into workspace" so that it will correctly refer to the nimble_peripheral_utils dir`

But it's not described this way in the official readme.

And the bleprph example still leads to a `Guru Meditation Error: Core  0 panic'ed (Store access fault). Exception was unhandled`... I remember that I already used this example with a previous installation and it worked like a charm.

Hoiss, 2022-12-09:
OK, I removed everything, went back to eclipse 2022-06 and installed ESP-IDF 5 release again.
Using the workaround with the entry to the CMakeLists.txt solves the problem, but the example still ends in an exception after sending one BLE packet.
Anyway, I could setup my own project, it compiles now and also works as assumed.

Somehow the integration to eclipse feels a bit error-prone to me.

Thanks your your help

valkuc, 2023-03-20:
I just installed ```espressif-ide-setup-2.8.1-with-esp-idf-5.0``` and having same issue with Hello World sample. I copied it into workspace, selected my build target (esp32-s2). Compilation performed successfully, but Eclipse IDE shows unresolved defines. Tried reindex/refresh - nothing helped.
[image]

GTCLive, 2023-04-13:
> I just installed `espressif-ide-setup-2.8.1-with-esp-idf-5.0` and having same issue with Hello World sample. I copied it into workspace, selected my build target (esp32-s2). Compilation performed successfully, but Eclipse IDE shows unresolved defines. Tried reindex/refresh - nothing helped. [image]

Move esp_chip_info.h to the top, like so:

#include "esp_chip_info.h"
#include <stdio.h>
#include <inttypes.h>
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_flash.h"