espressif/idf-eclipse-plugin #654: Error while building a template program

Mroogal, 2022-12-10:
**Describe the bug**
Hi,
I tried to build and lunch a template program on my ESP32-S3 dev-kit and problem occured. It has a problem with building and shows error:

Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'blink'.
'int com.espressif.idf.core.build.IDFBuildConfiguration.watchProcess(java.lang.Process, org.eclipse.cdt.core.resources.IConsole)'

I tried to follow the tutorial from: [https://github.com/espressif/idf-eclipse-plugin#cmakeproject](https://github.com/espressif/idf-eclipse-plugin#cmakeproject) to setup IDE but maybe I lost some steps in-between while configuring a workplace.

**To Reproduce**
Steps to reproduce the behavior:
1. Build template program.

**Expected behavior**
Expected behaviour is being able to build project properly.

**Screenshots**
[image]

**Eclipse Error log:** 
I attached a file with a log.
[blink_project_build_error_log.txt](https://github.com/espressif/idf-eclipse-plugin/files/10201170/blink_project_build_error_log.txt)

kolipakakondal, 2022-12-11:
Hi @Mroogal i am sorry for the issue. Looks like you're using the latest Eclipse version `Eclipse 2022-12` which is not yet supported by the plugin. I would recommend to use the `Eclipse 2022-09` version which is more stable and support IDF Eclipse plugin

You can find 2022-09 releases here https://www.eclipse.org/downloads/packages/release/2022-09/r

Anyway thanks for reporting this issue, will look into it.

Coffeeye, 2022-12-12:
Can confirm, latest update on 2022-12 cause error.
Rollback changes or redownload version 2022-12 seems to have no effect.
Changing to 2022-09 solves the issue.

```
Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project.
'int com.espressif.idf.core.build.IDFBuildConfiguration.watchProcess(java.lang.Process, org.eclipse.cdt.core.resources.IConsole)'
```

Mroogal, 2022-12-12:
Yeah, I changed back to 2022-09 and it worked, so the issue is with 2022-12 version of the eclipse.

kolipakakondal, 2022-12-13:
Thanks @Mroogal @Coffeeye for verifying and updating on this.

Nike-Prallow, 2023-04-06:
Thank you! Now version CDT 2023-03 works.

kolipakakondal, 2023-04-12:
As [Eclipse 2022-12](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1) supported is added with v2.9.1 build, hence closing this issue.

closed 2023-04-12 by kolipakakondal