espressif/idf-eclipse-plugin #666: Eclipse Espresif Idf plugin failes

jbhome, 2022-12-19:
I have tried both linux mint and ubuntu 2204 to install eclipse and the esp-idf plugin.
Using the Espresif install instructions, things work, up to the point where one starts creating a new project.
At this point, you refer to various menu items present in the launch bar.
Unfortunately, Eclipse has dropped this from the current release. There are some instructions referring to the GDB Launch/Launch menu, and enabling the Launch bar however this item the enable Launch tool bar sub-item under the GDB launch item is not present.
After poking around for quite some time, I did get this item visible and was able to turn the Launch bar on but in subsequent attempts, I've been unable to find how to do it.
The reason I'm looking at the Eclipse plugin is the graphical benefits during debugging.
a
**To Reproduce**
Steps to reproduce the behavior:
1. Install prerequisites
2. Install Eclipse
3. Install ESP-Idf plugin 
4. Follow instructions to the point where it says click the third dropdown window (no pictures!?) and select New Launch Target.

**Expected behavior**
At this point, you are expecting the Launch toolbar to be visible which it is not.
People have worked around this using some shell commands and some eclipse functions but the whole point is one should be able to follow the instructions and get the desired results. After all the goal is not to get in depth knowledge of the inner workings of Eclipse but rather to design, code, build, flash and debug an ESP32 projects.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
Espressif > Product Information (Copy content from the console and attach as a file)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here [](https://github.com/espressif/idf-eclipse-plugin#error-log) 
[Espressif_IDF_Plugin.txt](https://github.com/espressif/idf-eclipse-plugin/files/10250321/Espressif_IDF_Plugin.txt)
[Eclipse_error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/10250346/Eclipse_error_log.log)
[Preferences_View.pdf](https://github.com/espressif/idf-eclipse-plugin/files/10263328/Preferences_View.pdf)

AndriiFilippov, 2022-12-20:
Hi @jbhome  ! 

I have tested your issue under: 
OS: ubuntu 22.04
ESP-IDF: v5.0 
Eclipse: 22-09 
ESP Eclipse Plug-in: v2.7 

LGTM. 

Please, make sure you install all the prerequisites properly.

1) Install prerequisites
2) Install Eclipse
3) Install ESP-Idf plugin
4) Espressif -> Download and Configure ESP-IDF -> Select your ESP-IDF or download one -> Finish tools installation -> Create new ESP-IDF project using Project Explorer window(left side) -> click the third dropdown window and select New Launch Target.

"Follow instructions to the point where it says click the third dropdown window (no pictures!?) and select New Launch Target. "
Please, see Third dropdown window below: 

[image]

jbhome, 2022-12-20:
Thanks for the quick response but...
This the problem.
I followed all steps but my eclipse does not have that bar enabled. The
picture is sent show there is no enable option for the launch bar. This is
the problem I have.
What versions of eclipse and the idf are you using?
I have tried this  from git, ubunto software installer eclipse market place
all with same results.

jbhome, 2022-12-20:
Missed the versions.
I will try it again. Don't know what I'm doing wrong.

kolipakakondal, 2022-12-21:
> !SESSION 2022-12-05 17:15:21.106 -----------------------------------------------
eclipse.buildId=4.25.0.I20220831-1800
java.version=17.0.4.1
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=en_CA
Framework arguments:  -product org.eclipse.epp.package.java.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.java.product

Hello @jbhome, You're using Eclipse CDT 2022-12 version which is not supported by the plugin. Kindly use the Eclipse CDT 2022-09.

https://www.eclipse.org/downloads/packages/release/2022-09/r/eclipse-ide-cc-developers

sigmaaa, 2022-12-21:
Hi @jbhome, 

Have you tried to use Espressif-IDE? It has a plugin and the eclipse prerequisites https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.7.0-linux.gtk.x86_64.tar.gz

kolipakakondal, 2022-12-22:
Duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/654

McTrk, 2023-03-02:
> Thanks for the quick response but... This the problem. I followed all steps but my eclipse does not have that bar enabled. The picture is sent show there is no enable option for the launch bar. This [...]

I was able to make the Launch Toolbar appear on macOS Eclipse 4.7 by installing LaunchBar:
[image]

kolipakakondal, 2023-03-02:
Hi @McTrk Launchbar plugin `11.1.0` shows that your eclipse is updated with the Eclipse CDT 2022-12 plugins which is not yet supported by the IDF Eclipse plugin. Not sure how did you land there.

Could you please stick to Eclipse CDT 2022-09 version for compatibility.

If you're a windows user - pick Espressif-IDE Windows offline installer 

Espressif-IDE v2.8.1 and ESP-IDF 4.4.4 [download](https://github.com/espressif/idf-installer/releases/download/espressif-ide-2.8.1-esp-idf-4.4.4/espressif-ide-setup-2.8.1-with-esp-idf-4.4.4.exe)/[mirror](https://dl.espressif.com/dl/idf-installer/espressif-ide-setup-2.8.1-with-esp-idf-4.4.4.exe)

Espressif-IDE v2.8.1 and ESP-IDF 4.4.4 [download](https://github.com/espressif/idf-installer/releases/download/espressif-ide-2.8.1-esp-idf-4.4.4/espressif-ide-setup-2.8.1-with-esp-idf-4.4.4.exe)/[mirror](https://dl.espressif.com/dl/idf-installer/espressif-ide-setup-2.8.1-with-esp-idf-4.4.4.exe)

macOS user
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v2.8.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v2.8.1.dmg

Linux user
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.8.1-linux.gtk.x86_64.tar.gz

These packages come with the Eclipse CDT 2022-09 so you won't have the issue reported above.

kolipakakondal, 2023-03-02:
> > Errors running builder 'CDT Core Builder' on project 'hello'.
> > 'int com.espressif.idf.core.build.IDFBuildConfiguration.watchProcess(java.lang.Process, org.eclipse.cdt.core.IConsoleParser[])'

Hi @McTrk If you've reproducible steps how did you land there with Espressif-IDE v2.8.0/2.8.1/Eclipse CDT 2022-09, please help us. We can take a look.

Hi @AndriiFilippov  Did you come across this error while working with the Espressif-IDE v2.8.0/2.8.1/Eclipse CDT 2022-09?  We can check internally.

McTrk, 2023-03-03:
Thanks for the reply. FYI, I landed there by installing both CDT and ESP IDF plugin from Eclipse Marketplace.

> Hi @McTrk Launchbar plugin `11.1.0` shows that your eclipse is updated with the Eclipse CDT 2022-12 plugins which is not yet supported by the IDF Eclipse plugin. Not sure how did you land there.
> 
> Could you please stick to Eclipse CDT 2022-09 version for compatibility.
[...]
> These packages come with the Eclipse CDT 2022-09 so you won't have the issue reported above.
Understood. I'll see if I can tweak the download sites and downgrade CDT.

kolipakakondal, 2023-04-12:
Eclipse CDT 2012-12 support is added with the [v2.9.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1) release, hence closing this issue.

closed 2023-04-12 by kolipakakondal