espressif/idf-eclipse-plugin #668: IDF Eclipse Plugin installs all ESP-IDF tools when only a subset was originally install by ESP-IDF installer

JBHunterI, 2022-12-22:
**Describe the bug**
Plugin's Espressif -> ESP-IDF Tools Manager -> Install Tools blindly installs all tools even though, as requested, (v2.18-beta1) ESP-IDF Installer had only installed a subset of these. 

**To Reproduce**
Steps to reproduce the behavior:
1. Click Plugin's Espressif -> ESP-IDF Tools Manager -> Install Tools to add required variables to Eclipse Preferences -> C/C++ ->Build -> Environment (and whatever else Plugin requires)
2. Observe the first thing that appears in the Espressif IDF Tools Console is:
Installing tools...
This can take a while. Please be patient.
Executing %IDF_TOOLS_PATH%\python_env\idf4.4_py3.8_env\Scripts\python.exe %IDF_PATH%\tools\idf_tools.py install _**all**_

**Expected behavior**
Makes no sense that the tools need to be re-installed when Plugin was told to "Use an existing ESP-IDF directory from file system" in  Espressif -> Download & Configure ESP-IDF, and ESP-IDF Directory was specified in Espressif -> ESP-IDF Tools Manager -> Install Tools, but if they must be re-installed then the first thing that should appear in the Espressif IDF Tools Console should be:
Installing tools...
This can take a while. Please be patient.
Executing %IDF_TOOLS_PATH%\python_env\idf4.4_py3.8_env\Scripts\python.exe %IDF_PATH%\tools\idf_tools.py install _**[previously installed tools]**_ ...

or better yet:
**Detected ESP-IDF already installed, skipping tools installation.**

Honestly, the whole tools installation thing within the Plugin needs to be looked at again. I made a batch file that starts Eclipse from a ESP-IDF CMD shell (installed during ESP-IDF installation) before closing the CMD shell so the Plugin would have all that (environment) information available, but the Plugin just seems to ignore it. I simply added this near the end of ESP-IDF's idf_cmd_init.bat:
```
...
REM personal customization
set IDF_TARGET=esp32c3
set ESPPORT=COM8
set ESPBAUD=115200
if [%IdfCmdInit_startEclipse%] NEQ [] (
    set IdfCmdInit_startEclipse=
    start C:\App\eclipseESP-IDF\eclipse.exe && exit 0
)
:: Check if this is a recent enough copy of ESP-IDF.
:: If so, use export.bat provided there.
:: Note: no "call", will not return into this batch file.
if exist "%IDF_PATH%\export.bat" %IDF_PATH%\export.bat
```

The above IdfCmdInit_startEclipse variable is set in a eclipse.cmd that I can launch from Windows' Start Menu, like so:
```
@echo off
REM This file goes into %USERPROFILE%\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\
REM and is called by a  %USERPROFILE%\AppData\Roaming\Microsoft\Windows\Start Menu\Programs\EclipseESP-IDF.lnk" shortcut
SETLOCAL ENABLEDELAYEDEXPANSION ENABLEEXTENSIONS
REM Find & run "ESP-IDF vX.Y CMD.lnk" as installed in "Windows Start -> ESP-IDF" by ESP-IDF Installer.
cd "%APPDATA%""\Microsoft\Windows\Start Menu\Programs\ESP-IDF"
set IdfCmdInit_startEclipse=1
FOR /F "tokens=* USEBACKQ" %%F IN (`dir /b *CMD.lnk`) DO ("%%F")
```
The Product Information seems to be fine with that, ending in:
...
Operating System: windows 10
Java Runtime Version: 17.0.4.1+1
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.7.0.202211101524
ESP-IDF v4.4.3
Python set for IDF_PYTHON_ENV: Python 3.8.7

Seems that would be the right way to configure the Plugin since it ultimately depends on the installed ESP-IDF. Why redo the whole ESP-IDF installation again from within Eclipse?

**Screenshots**
ESP-IDF Online Installer v2.18-beta3 has corrected related problems (see [CMD and PS shells complain of an error if installtion intentionally omitted support for even one ESP32 target (IDFGH-8879) #10300](https://github.com/espressif/esp-idf/issues/10300))

kolipakakondal, 2022-12-23:
Hi @JBHunterI Thanks for sharing the detailed feedback and pain points. Appreciate for helping us to improve the tooling.

Let me give you a couple of pointers here:

1. If you're using ESP-IDF Windows Online or Offline installer, you don't need to run the `Install Tools` process again from the IDE.  All the required build environment variables are will be auto-configured by the plugin- for example, configuring IDF_PATH, PATH, and other required variables. I have mentioned [here](https://medium.com/the-esp-journal/espressif-ide-and-whats-new-in-v2-4-0-part-2-e3b6c8de185c), probably need to add this in the plugin documentation as well.
2. But if you run  `Espressif -> ESP-IDF Tools Manager -> Install Tools` from Eclipse, all the tools will be installed if it's not installed already.
3. Plugin knows only about the esp_idf.json from the installation folder (.espressif/esp_idf.json) for reading particular variables- gitPath, idfToolsPath, and idfinstalled. Based on these variables it will configure the build environment variables as soon as the IDE is loaded. 
4.  Plugin doesn't know anything about the idf_cmd_init.bat file so it ignores all variables you set there
5. I think one important point which you brought up here - As the installation started with the ESP-IDF Installer, the plugin should know what is already installed. For example, what were the targets selected during the installation so that plugin does needn't need to install everything again? For the same reason, we have introduced `Espressif -> ESP-IDF Tools Manager -> Tools Installation Wizard' where you should be able to see what's already installed(that's based on the folders it found) and choose to install other tools if required. Going forward will replace [Tools Installation Wizard](https://github.com/espressif/idf-eclipse-plugin#InstallToolsWizard) with the `Install Tools` wizard. Any feedback on this would be a great help as well.

alirana01, 2023-03-20:
The issue will be fixed with the latest release