espressif/idf-eclipse-plugin #670: Application Size Analysis shows error in Espressif-IDE

prookyon, 2023-01-02:
**Describe the bug**
Application Size Analysis shows error in Espressif-IDE 2.7.0 with message "Failed to create the part's controls"
All following info is from a totally clean install in virtual environment.
```
java.lang.NullPointerException
	at com.espressif.idf.ui.size.IDFSizeOverviewComposite.createPartControl(IDFSizeOverviewComposite.java:58)
	at com.espressif.idf.ui.size.IDFSizeAnalysisEditor.createOverviewPage(IDFSizeAnalysisEditor.java:121)
	at com.espressif.idf.ui.size.IDFSizeAnalysisEditor.createPages(IDFSizeAnalysisEditor.java:52)
	at org.eclipse.ui.part.MultiPageEditorPart.createPartControl(MultiPageEditorPart.java:333)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.createPartControl(CompatibilityPart.java:158)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityEditor.createPartControl(CompatibilityEditor.java:96)
	at org.eclipse.ui.internal.e4.compatibility.CompatibilityPart.create(CompatibilityPart.java:365)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:566)
	at org.eclipse.e4.core.internal.di.MethodRequestor.execute(MethodRequestor.java:58)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:995)
	at org.eclipse.e4.core.internal.di.InjectorImpl.processAnnotated(InjectorImpl.java:960)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalInject(InjectorImpl.java:140)
	at org.eclipse.e4.core.internal.di.InjectorImpl.internalMake(InjectorImpl.java:403)
	at org.eclipse.e4.core.internal.di.InjectorImpl.make(InjectorImpl.java:330)
	at org.eclipse.e4.core.contexts.ContextInjectionFactory.make(ContextInjectionFactory.java:202)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.createFromBundle(ReflectionContributionFactory.java:91)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.doCreate(ReflectionContributionFactory.java:60)
	at org.eclipse.e4.ui.internal.workbench.ReflectionContributionFactory.create(ReflectionContributionFactory.java:42)
	at org.eclipse.e4.ui.workbench.renderers.swt.ContributedPartRenderer.createWidget(ContributedPartRenderer.java:132)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createWidget(PartRenderingEngine.java:995)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:659)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.safeCreateGui(PartRenderingEngine.java:763)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine$2.run(PartRenderingEngine.java:728)
	at org.eclipse.core.runtime.SafeRunner.run(SafeRunner.java:45)
	at org.eclipse.e4.ui.internal.workbench.swt.PartRenderingEngine.createGui(PartRenderingEngine.java:712)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.lambda$0(PartServiceImpl.java:105)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler.lambda$0(UIEventHandler.java:38)
	at org.eclipse.swt.widgets.Synchronizer.syncExec(Synchronizer.java:183)
	at org.eclipse.ui.internal.UISynchronizer.syncExec(UISynchronizer.java:133)
	at org.eclipse.swt.widgets.Display.syncExec(Display.java:4792)
	at org.eclipse.e4.ui.workbench.swt.DisplayUISynchronize.syncExec(DisplayUISynchronize.java:34)
	at org.eclipse.e4.ui.services.internal.events.UIEventHandler.handleEvent(UIEventHandler.java:38)
	at org.eclipse.equinox.internal.event.EventHandlerWrapper.handleEvent(EventHandlerWrapper.java:205)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:203)
	at org.eclipse.equinox.internal.event.EventHandlerTracker.dispatchEvent(EventHandlerTracker.java:1)
	at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:234)
	at org.eclipse.osgi.framework.eventmgr.ListenerQueue.dispatchEventSynchronous(ListenerQueue.java:151)
	at org.eclipse.equinox.internal.event.EventAdminImpl.dispatchEvent(EventAdminImpl.java:133)
	at org.eclipse.equinox.internal.event.EventAdminImpl.sendEvent(EventAdminImpl.java:75)
	at org.eclipse.equinox.internal.event.EventComponent.sendEvent(EventComponent.java:44)
	at org.eclipse.e4.ui.services.internal.events.EventBroker.send(EventBroker.java:55)
	at org.eclipse.e4.ui.internal.workbench.UIEventPublisher.notifyChanged(UIEventPublisher.java:63)
	at org.eclipse.emf.common.notify.impl.BasicNotifierImpl.eNotify(BasicNotifierImpl.java:424)
	at org.eclipse.e4.ui.model.application.ui.impl.ElementContainerImpl.setSelectedElementGen(ElementContainerImpl.java:170)
	at org.eclipse.e4.ui.model.application.ui.impl.ElementContainerImpl.setSelectedElement(ElementContainerImpl.java:188)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.showElementInWindow(ModelServiceImpl.java:660)
	at org.eclipse.e4.ui.internal.workbench.ModelServiceImpl.bringToTop(ModelServiceImpl.java:624)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.delegateBringToTop(PartServiceImpl.java:787)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.bringToTop(PartServiceImpl.java:401)
	at org.eclipse.e4.ui.internal.workbench.PartServiceImpl.showPart(PartServiceImpl.java:1261)
	at org.eclipse.ui.internal.WorkbenchPage.busyOpenEditor(WorkbenchPage.java:3210)
	at org.eclipse.ui.internal.WorkbenchPage.lambda$9(WorkbenchPage.java:3100)
	at org.eclipse.swt.custom.BusyIndicator.showWhile(BusyIndicator.java:74)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3098)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3068)
	at org.eclipse.ui.internal.WorkbenchPage.openEditor(WorkbenchPage.java:3051)
	at org.eclipse.ui.ide.IDE.openEditor(IDE.java:347)
	at com.espressif.idf.ui.size.IDFSizeMemoryHandler$1.run(IDFSizeMemoryHandler.java:106)
	at org.eclipse.swt.widgets.RunnableLock.run(RunnableLock.java:40)
	at org.eclipse.swt.widgets.Synchronizer.runAsyncMessages(Synchronizer.java:132)
	at org.eclipse.swt.widgets.Display.runAsyncMessages(Display.java:4043)
	at org.eclipse.swt.widgets.Display.readAndDispatch(Display.java:3648)
[... 81 lines cut ...]
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.0_py3.8_env

Operating System: windows 10
Java Runtime Version: 11.0.17+8-LTS
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.7.0.202211101524
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule vendor/c_exception
fatal: 'git status --porcelain=2' failed in submodule components/cmock/CMock
WARNING: Git version unavailable, reading from source
ESP-IDF v5.0.0
Python set for IDF_PYTHON_ENV: Python 3.8.7
```

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/10329834/error_log.log)

kolipakakondal, 2023-01-02:
Hi @prookyon 

Thanks for reporting. We have already made the [fix](https://github.com/espressif/idf-eclipse-plugin/pull/642), and it will be available in the next release.

If you would like to try the development build, the change is part of the nightly update site https://dl.espressif.com/dl/idf-eclipse-plugin/updates/nightly/

Please verify and let us know if that helps.

prookyon, 2023-01-02:
Thanks, I can see this works when using the nightly update site with Eclipse.
Although on Espressif-IDE (2.7 from the online installer) adding that site and installing the nightly IDF plugin results in a very broken installation (many Eclipse components stop working). I guess the way Espressif-IDE is bundled it's necessary to wait for the official release? Added the error log also just in case it's useful.
[error_log_2.log](https://github.com/espressif/idf-eclipse-plugin/files/10332068/error_log_2.log)

kolipakakondal, 2023-01-03:
> Thanks, I can see this works when using the nightly update site with Eclipse. Although on Espressif-IDE (2.7 from the online installer) adding that site and installing the nightly IDF plugin results [...]

Thanks @prookyon for updating on this.

Looks like the broken installation is because of the latest Eclipse version(Eclipse 2022-12). IDF Eclipse plugin/Espressif-IDE [doesn't support this yet](https://github.com/espressif/idf-eclipse-plugin/issues/654) and also new version of eclipse have a dependency on Java 17. 

Please continue to use Eclispe CDT 2022-09 with the IEP plugin 2.7.0 or Espressif-IDE v2.7.0. On both of them nightly build should work fine as long as you're not updating the Eclipse base version.

We are working on Eclipse 2022-12 support, will keep it posted.

I hope this clarifies.

kolipakakondal, 2023-02-07:
Application Size Analysis shows error in Espressif-IDE is fixed with the v2.8.0 release. Hence closing the issue

closed 2023-02-07 by kolipakakondal