espressif/idf-eclipse-plugin #672: V2.7.0 doesn't correctly install IDF-V5.0 offline tools

DspDave, 2023-01-03:
**Describe the bug**
In Windows 10 using the Eclipse IDE, If the **offline** use of **_IDF-V4.4.3_** is updated to the **offline** **_IDF-V5.0_** the tools are not correctly installed.

**To Reproduce**
.1 Download the new ESP-IDF V5.0 installer (esp-idf-tools-setup-offline-5.0.exe) and the espressif-ide-setup-2.7.0-with-esp-idf-4.4.3.exe.
.2 Disconnect the internet connection.
.3 Run espressif-ide-setup-2.7.0-with-esp-idf-4.4.3.exe to install Eclipse IDE & ESP-IDF V4.4.3 tools, can now build 'blink' project OK.
.4 Run new esp-idf-tools-setup-offline-5.0.exe which completes by running idf-command-init.bat & install.sh without issue.
.5 Using the Espressif->Download and Configure IDF & selecting the 'C:\Espressif\frameworks\esp-idf-v5.0'
.6 Close Eclipse & re-open Eclipse, to be sure...
.7 Delete Build Folder & rebuild project
.8 Project fails to build with following errors (see BuildFails001.txt)
.9 Use the Espressif->ESP-IDF Tools Manager->Install Tool option to update the IDF & tools launches the same 'Install tools' dialog and ends in the same result.
.10 Use the Espressif->ESP-IDF Tools Manager->Tools Installation Wizard to update the IDF & tools shows all of the V5.0 tools as uninstalled & generates the following log (see InstallToolsWizard.txt).

Eclipse Error Logs have been captured whilst using the different plugin IDF update methods.

**Expected behavior**
The active IDF & Tool chain tool chain should use IDF-V5.0.

**Unexpected Behaviour**
Again if the internet is connected this works, however the values configured in 'espidf.constraints.v5.0.txt' seem incorrect as the IDF actually has library scripts in it that are too new?

The line in the build log **'Are you connected to the internet?'** seems suspicious for off-line operation.

Attempting to list installed tools launches the dialog asking for tool installation again?

**Screenshots**
[ErrorInfo.zip](https://github.com/espressif/idf-eclipse-plugin/files/10338330/ErrorInfo.zip)
[image]

**Espressif-IDE Product Information:**
[EspressifProductInfo.txt](https://github.com/espressif/idf-eclipse-plugin/files/10338324/EspressifProductInfo.txt)

**Eclipse Error log:** 
[ErrorLogs.zip](https://github.com/espressif/idf-eclipse-plugin/files/10338325/ErrorLogs.zip)

kolipakakondal, 2023-01-11:
Hi @DspDave Thanks for reporting and sharing the logs. Will take a look.

AndriiFilippov, 2023-01-12:
@kolipakakondal hi ! 
I was able to reproduce this one.

alirana01, 2023-01-24:
@DspDave thank you for reporting the issue we are working on a solution for this one