espressif/idf-eclipse-plugin #675: 改为中文后ide启动失败

OPEN-F, 2023-01-14:
**Describe the bug**
改为中文后ide启动失败，如下图

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Chinese or 中文'
2. 退出ide
3. See error

**Expected behavior**
ide不能打开，并弹出如下图报错

**Screenshots**
[image]

**Espressif-IDE Product Information:**
2.7.0

**Eclipse Error log:** 
详情查看附件

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log 
[1673711154074.log](https://github.com/espressif/idf-eclipse-plugin/files/10418094/1673711154074.log)

sigmaaa, 2023-01-17:
Hi @OPEN-F,

Looks like you are using java 8, but Java 11 is required. Because of Java 8 a lot of requirements are unresolved.
Download and install Java SE from [here](https://www.oracle.com/technetwork/java/javase/downloads/index.html)

OPEN-F, 2023-01-17:
非常感谢您的帮助，但是在我安装Java11后出现了如下的问题
[image]
[image]
我也尝试过安装最新的java17，但依然出现了类似的问题，如下图
[image]
[image]

OPEN-F, 2023-01-17:
@sigmaaa 非常感谢你的帮助，但是在我安装了Java11后（卸载了java8）,又重新安装了Espressif-IDE，出现了如下问题（前两幅图片）
[image]
[image]
我用同样的方法安装了新版的java17，也出来类似问题。（后两幅图片）
[image]
[image]

OPEN-F, 2023-01-17:
[image]
这是JAVA的版本号

sigmaaa, 2023-01-17:
Hi @OPEN-F,

Maybe the problem is related to the default JVM path. Can you check your espressif-ide.ini file in the Espressif IDE folder? Try to edit it with `-vm` argument like this:
```
-startup
plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
--launcher.library
plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.2.600.v20220720-1916
-data
@noDefault
-vm 
C:\Program Files\Java\jdk-11.0.9\bin\server\jvm.dll
-vmargs
-Xms256m
-Xmx2048m
-Dosgi.requiredJavaVersion=11
-Dosgi.instance.area.default=@user.home/espressif-workspace
```

where instead of `C:\Program Files\Java\jdk-11.0.9\bin\server\jvm.dll` should path relative to your Java folder.

OPEN-F, 2023-01-17:
Hi @sigmaaa 
在你的帮助下，我更改了ini文件（如下）,使ide顺利启动，但是仍然不能使ide的语言更改为中文，如下图所示，这是什么原因呢？
```
-startup
plugins/org.eclipse.equinox.launcher_1.6.400.v20210924-0641.jar
--launcher.library
plugins/org.eclipse.equinox.launcher.win32.win32.x86_64_1.2.600.v20220720-1916
-vm 
D:\program files\Java\jdk-11.0.15\bin\server\jvm.dll
-nl
zh
-data
@noDefault
-vmargs
-Dosgi.instance.area.default=@user.home/espressif-workspace
-Dosgi.requiredJavaVersion=11
-Xmx2048m
-Xms256m
```

OPEN-F, 2023-01-17:
你好，我想对我们交流的问题做一个总结教程并在一些开放性平台上分享，其中可能会涉及到您告知我的解决方法，可以吗？

closed 2023-01-18 by sigmaaa