espressif/idf-eclipse-plugin #679: ESP32-S3 FreeRTOS Real time statistics using FreeRTOS plugin

krupis, 2023-01-20:
Hello. I have initially created an issue on esp-idf repository:
https://github.com/espressif/esp-idf/issues/10101
but it was recommended to put it up here.

I simply want to know if there is a way to monitor ESP32-S3 device in real time and check task status, memory usage, priorities and etc...
I can see information such as this:
[image]

but only when I place a breakpoint. Is there a way to get this for the FreeRTOS (task list) to update in real time without placing breakpoints?

ChrisJP84, 2023-01-20:
+1. Seems a fair few people would like this.

kolipakakondal, 2023-01-23:
Hi @krupis The plugin doesn't have this feature yet to get real-time data. Will consider this in our future releases. Thanks for your request.

GTCLive, 2023-02-03:
Hello @kolipakakondal - I personally wouldn't make it a consideration, but a requirement rather. Quite an essential (vital?) debugging asset. Appreciate your continuous support and efforts.

lucamae, 2024-06-05:
Hi, I tried the same and for me would be enough to see task info when stopped at breakpoint, but I cannot see nothing.
[image]
Should I need to configure something else? I already set CONFIG_FREERTOS_USE_TRACE_FACILITY=y
(esp-idf 5.0.6)