espressif/idf-eclipse-plugin #682: Update ESP-IDF removes Plugin

franz-ms-muc, 2023-01-27:
**Describe the bug**
Update ESP-IDF removes Plugin

**To Reproduce**
Steps to reproduce the behavior:
1. espressif-ide-setup-2.7.0-with-esp-idf-5.0.exe
2. install normally
3. all OK
4. Go to "install new Software" select the Espressif Plugin, Update. 
5. Restart
6. Espressif Plugin is missing; no "Espressif" Menu on the Top.

sigmaaa, 2023-01-27:
Hi @franz-ms-muc,
Thank you for reporting this. Probably, during the update, the plugin dependencies like eclipse CDT also updated and the new version of dependencies requires Java 17. This might be the problem, that dependencies are not getting resolved. 

Please, attach the [error log](https://github.com/espressif/idf-eclipse-plugin#error-log) if you have one, this will help. But it's ok if you don't have it, I will try to reproduce it later today.

franz-ms-muc, 2023-01-27:
[image]

franz-ms-muc, 2023-01-27:
[image]

franz-ms-muc, 2023-01-27:
> Please, attach the [error log](https://github.com/espressif/idf-eclipse-plugin#error-log) if you have one, this will help. But it's ok if you don't have it, I will try to reproduce it later today.

The Error Log seems gone as well.

franz-ms-muc, 2023-01-27:
if i download 
eclipse-embedcpp-2022-09-R-win32-x86_64.zip
and install the Plugin [plugin](https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url)
then it in fact show massive Information:

[image]

[image]

franz-ms-muc, 2023-01-27:
i will try the 2nd option: 
[image]

franz-ms-muc, 2023-01-27:
All OK on a Eclipse. 

[image]

kolipakakondal, 2023-04-12:
Eclipse 2022-12 support is added with [v2.9.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1), hence closing this issue

closed 2023-04-12 by kolipakakondal