espressif/idf-eclipse-plugin #683: Serial Port gets mixed up in Linux

f-hoepfinger-hr-agrartechnik, 2023-01-27:
Eclipse IDE for C/C++ Developers (includes Incubating components)
Version: 2022-09 (4.25.0)

ESP-IDF Eclipse Plugin
Version: 2.7.0.202211101524
Copyright (c) 2021 ESPRESSIF SYSTEMS (SHANGHAI) CO., LTD. All rights reserved.

No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.1 LTS
Release:	22.04
Codename:	jammy

Above Configuration leads to the Following Problem: 

when the ESP-IDF Serial Monitor is open, and stay open during flashing, 
it does not reconnect after flashing. 

in Windows i cannot see this Problem.

f-hoepfinger-hr-agrartechnik, 2023-01-27:
device reports readiness to read but returned no data (device disconnected or multiple access on port?)  
Waiting for the device to reconnect0;33m

f-hoepfinger-hr-agrartechnik, 2023-01-27:
Executing action: flash  
Running ninja in directory /home/franz/git/hr/CTIS2/UnitTest/build  
Executing "ninja flash"...  
\[1/5\] cd /home/franz/git/hr/CTIS2/UnitTest/build/esp-idf/esptool\_py && python /home/franz/git/hr/esp-idf/components/partition\_table/check\_sizes.py --offset 0x8000 partition --type app /home/franz/git/hr/CTIS2/UnitTest/build/partition\_table/partition-table.bin /home/franz/git/hr/CTIS2/UnitTest/build/unit\_test\_test.bin  
unit\_test\_test.bin binary size 0x9f880 bytes. Smallest app partition is 0x200000 bytes. 0x160780 bytes (69%) free.  
\[2/5\] Performing build step for 'bootloader'  
\[1/1\] cd /home/franz/git/hr/CTIS2/UnitTest/build/bootloader/esp-idf/esptool\_py && python /home/franz/git/hr/esp-idf/components/partition\_table/check\_sizes.py --offset 0x8000 bootloader 0x1000 /home/franz/git/hr/CTIS2/UnitTest/build/bootloader/bootloader.bin  
Bootloader binary size 0x6df0 bytes. 0x210 bytes (2%) free.  
\[2/3\] cd /home/franz/git/hr/esp-idf/components/esptool\_py && /home/franz/.espressif/tools/cmake/3.24.0/bin/cmake -D IDF\_PATH=/home/franz/git/hr/esp-idf -D "SERIAL\_TOOL=python;;/home/franz/git/hr/esp-idf/components/esptool\_py/esptool/esptool.py;--chip;esp32" -D "SERIAL\_TOOL\_ARGS=--before=default\_reset;--after=hard\_reset;write\_flash;@flash\_args" -D WORKING\_DIRECTORY=/home/franz/git/hr/CTIS2/UnitTest/build -P /home/franz/git/hr/esp-idf/components/esptool\_py/run\_serial\_tool.cmake  
esptool esp32 -p /dev/ttyUSB4 -b 460800 --before=default\_reset --after=hard\_reset write\_flash --flash\_mode dio --flash\_freq 80m --flash\_size 8MB 0x1000 bootloader/bootloader.bin 0x20000 unit\_test\_test.bin 0x8000 partition\_table/partition-table.bin 0xf000 ota\_data\_initial.bin  
esptool.py v4.4  
Serial port /dev/ttyUSB4  
Connecting..........

A serial exception error occurred: Could not configure port: (5, 'Input/output error')  
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.  
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html  
CMake Error at run\_serial\_tool.cmake:55 (message):  
   
 python;;/home/franz/git/hr/esp-idf/components/esptool\_py/esptool/esptool.py;--chip;esp32  
 failed

  
FAILED: CMakeFiles/flash /home/franz/git/hr/CTIS2/UnitTest/build/CMakeFiles/flash   
cd /home/franz/git/hr/esp-idf/components/esptool\_py && /home/franz/.espressif/tools/cmake/3.24.0/bin/cmake -D IDF\_PATH=/home/franz/git/hr/esp-idf -D "SERIAL\_TOOL=python;;/home/franz/git/hr/esp-idf/components/esptool\_py/esptool/esptool.py;--chip;esp32" -D "SERIAL\_TOOL\_ARGS=--before=default\_reset;--after=hard\_reset;write\_flash;@flash\_args" -D WORKING\_DIRECTORY=/home/franz/git/hr/CTIS2/UnitTest/build -P /home/franz/git/hr/esp-idf/components/esptool\_py/run\_serial\_tool.cmake  
ninja: build stopped: subcommand failed.  
ninja failed with exit code 1, output of the command is in the /home/franz/git/hr/CTIS2/UnitTest/build/log/idf\_py\_stderr\_output\_623605 and /home/franz/git/hr/CTIS2/UnitTest/build/log/idf\_py\_stdout\_output\_623605

f-hoepfinger-hr-agrartechnik, 2023-01-27:
Error Log Attatched

```
!SESSION 2023-01-27 10:39:52.297 -----------------------------------------------
eclipse.buildId=4.25.0.I20220831-1800
java.version=17.0.4.1
java.vendor=Eclipse Adoptium
BootLoader constants: OS=linux, ARCH=x86_64, WS=gtk, NL=de_DE
Framework arguments:  -product org.eclipse.epp.package.cpp.product
Command-line arguments:  -os linux -ws gtk -arch x86_64 -product org.eclipse.epp.package.cpp.product

This is a continuation of log file /home/franz/git/hr/CTIS2/eclipse-workspace/.metadata/.bak_1.log
Created Time: 2023-01-27 12:53:08.312

!ENTRY com.espressif.idf.sdk.config.core 1 0 2023-01-27 12:53:08.312
!MESSAGE Command to be executed: {"set":{"LOG_DEFAULT_LEVEL_INFO":true},"version":2}

!ENTRY com.espressif.idf.sdk.config.ui 1 0 2023-01-27 12:53:08.459
!MESSAGE Set LOG_DEFAULT_LEVEL_INFO
{"version": 2, "values": {"LOG_DEFAULT_LEVEL_INFO": true, "LOG_DEFAULT_LEVEL_DEBUG": false, "LOG_DEFAULT_LEVEL": 3, "LOG_MAXIMUM_LEVEL_DEBUG": false, "LOG_MAXIMUM_LEVEL": 3}, "ranges": {"LWIP_TCPIP_TASK_STACK_SIZE": [2048, 65536]}, "visible": {"LOG_MAXIMUM_LEVEL_DEBUG": true}}

!ENTRY com.espressif.idf.sdk.config.core 1 0 2023-01-27 12:53:09.896
!MESSAGE Command to be executed: {"set":{"LOG_TIMESTAMP_SOURCE_SYSTEM":true},"version":2}

!ENTRY com.espressif.idf.sdk.config.ui 1 0 2023-01-27 12:53:10.061
!MESSAGE Set LOG_TIMESTAMP_SOURCE_SYSTEM
{"version": 2, "values": {"LOG_TIMESTAMP_SOURCE_RTOS": false, "LOG_TIMESTAMP_SOURCE_SYSTEM": true}, "ranges": {}, "visible": {}}

!ENTRY com.espressif.idf.sdk.config.core 1 0 2023-01-27 12:53:12.155
!MESSAGE Command to be executed: {"set":{},"save":null,"version":2}

!ENTRY com.espressif.idf.sdk.config.ui 1 0 2023-01-27 12:53:12.209
!MESSAGE Server connection closed

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:54:04.935
!MESSAGE org.java_websocket.WebSocketImpl@1b52d16c is terminated!

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:54:14.533
!MESSAGE [/home/franz/.espressif/python_env/idf5.0_py3.10_env/bin/python, /home/franz/git/hr/esp-idf/tools/idf_monitor.py, -p, /dev/ttyUSB4, -b, 115200, --ws, ws://localhost:32935, /home/franz/git/hr/CTIS2/UnitTest/build/unit_test_test.elf]

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:54:14.535
!MESSAGE Server started!

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:54:14.535
!MESSAGE {PATH=/home/franz/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/franz/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/franz/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/franz/.espressif/tools/xtensa-esp32s2-elf/esp-2022r1-11.2.0/xtensa-esp32s2-elf/bin:/home/franz/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin:/home/franz/.espressif/tools/xtensa-clang/14.0.0-38679f0333/xtensa-esp32-elf-clang/bin:/home/franz/.espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin:/home/franz/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/franz/.espressif/tools/cmake/3.24.0/bin:/home/franz/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/bin:/home/franz/.espressif/tools/ninja/1.10.2/:/home/franz/.espressif/python_env/idf5.0_py3.10_env/bin:/home/franz/git/hr/esp-idf/tools:/home/franz/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/snap/bin, XAUTHORITY=/run/user/1000/gdm/Xauthority, INVOCATION_ID=2fe7e92d21894b4ab77afe462e82b637, XMODIFIERS=@im=ibus, XDG_DATA_DIRS=/usr/share/ubuntu:/usr/share/gnome:/usr/local/share/:/usr/share/:/var/lib/snapd/desktop, GDMSESSION=ubuntu, DBUS_SESSION_BUS_ADDRESS=unix:path=/run/user/1000/bus,guid=85e8b768bf5775ba1b95f9e763d15d1f, ESP_IDF_VERSION=5.0, XDG_CURRENT_DESKTOP=ubuntu:GNOME, OPENOCD_SCRIPTS=/home/franz/.espressif/tools/openocd-esp32/v0.11.0-esp32-20221026/openocd-esp32/share/openocd/scripts, JOURNAL_STREAM=8:56492, SESSION_MANAGER=local/franz-exone-WORKSTATION-4204HE:@/tmp/.ICE-unix/2195,unix/franz-exone-WORKSTATION-4204HE:/tmp/.ICE-unix/2195, USERNAME=franz, LOGNAME=franz, PWD=/home/franz/eclipse, MANAGERPID=1876, SHELL=/bin/bash, GNOME_DESKTOP_SESSION_ID=this-is-deprecated, IDF_COMPONENT_MANAGER=1, GTK_MODULES=gail:atk-bridge, DBUS_STARTER_ADDRESS=unix:path=/run/user/1000/bus,guid=85e8b768bf5775ba1b95f9e763d15d1f, SYSTEMD_EXEC_PID=2195, XDG_SESSION_DESKTOP=ubuntu, SSH_AGENT_LAUNCHER=gnome-keyring, SHLVL=0, OXYGEN_DISABLE_INNER_SHADOWS_HACK=1, QT_IM_MODULE=ibus, XDG_CONFIG_DIRS=/etc/xdg/xdg-ubuntu:/etc/xdg, LANG=de_DE.UTF-8, XDG_SESSION_TYPE=x11, DISPLAY=:1, XDG_SESSION_CLASS=user, LIBOVERLAY_SCROLLBAR=0, IDF_PATH=/home/franz/git/hr/esp-idf, GDK_CORE_DEVICE_EVENTS=1, GPG_AGENT_INFO=/run/user/1000/gnupg/S.gpg-agent:0:1, DESKTOP_SESSION=ubuntu, USER=franz, XDG_MENU_PREFIX=gnome-, GIO_LAUNCHED_DESKTOP_FILE_PID=554717, QT_ACCESSIBILITY=1, WINDOWPATH=2, SSH_AUTH_SOCK=/run/user/1000/keyring/ssh, IDF_MAINTAINER=1, PYTHONUNBUFFERED=1, GNOME_SHELL_SESSION_MODE=ubuntu, XDG_RUNTIME_DIR=/run/user/1000, IDF_PYTHON_ENV_PATH=/home/franz/.espressif/python_env/idf5.0_py3.10_env, DBUS_STARTER_BUS_TYPE=session, HOME=/home/franz}

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:54:14.535
!MESSAGE {PATH=/home/franz/.espressif/tools/xtensa-esp-elf-gdb/11.2_20220823/xtensa-esp-elf-gdb/bin:/home/franz/.espressif/tools/riscv32-esp-elf-gdb/11.2_20220823/riscv32-esp-elf-gdb/bin:/home/franz/.espressif/tools/xtensa-esp32-elf/esp-2022r1-11.2.0/xtensa-esp32-elf/bin:/home/franz/.espressif/tools/xt [... cut ...]
!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:58:09.633
!MESSAGE Connection Made to: 

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:58:09.633
!MESSAGE 127.0.0.1

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:58:18.129
!MESSAGE org.java_websocket.WebSocketImpl@15b02411 is terminated!

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:58:56.834
!MESSAGE org.java_websocket.WebSocketImpl@4a40bb6e is terminated!

!ENTRY com.espressif.idf.core 1 0 2023-01-27 12:58:59.393
!MESSAGE /home/franz/git/hr/esp-idf/tools/idf.py -p /dev/ttyUSB4 flash
```

f-hoepfinger-hr-agrartechnik, 2023-01-27:
In my Opinion the Problem must be searched inside Linux: 

every time the Serial Port is thrown up, i disconnect and Reconnect the ESP-PROG again, 

and IF it gets a new Port Number like /dev/ttyUSB1, /dev/ttyUSB2, /dev/ttyUSB3 and so on, 

just in the Moment i have /dev/ttyUSB5, then it works again for 1-2 flash processes.

f-hoepfinger-hr-agrartechnik, 2023-01-27:
i want find out if same happens on Command Line: with idf.py flash, monitor etc..

f-hoepfinger-hr-agrartechnik, 2023-01-27:
so, command line not this Problem. Very Stable.

sigmaaa, 2023-01-29:
Hi @f-hoepfinger-hr-agrartechnik, 
Thank you for reporting this. 

> when the ESP-IDF Serial Monitor is open, and stay open during flashing,
> it does not reconnect after flashing.

sounds similar to this issue: https://github.com/espressif/idf-eclipse-plugin/issues/638. 
Can you check[ this](https://github.com/espressif/idf-eclipse-plugin/issues/638#issuecomment-1365145991), please, and tell us if it fix the problem?

sigmaaa, 2023-02-02:
Hi @f-hoepfinger-hr-agrartechnik 

The fix for it should be available in 2.8.0 also https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.8.0

f-hoepfinger-hr-agrartechnik, 2023-02-08:
just installing 2.8.1 and testing !

f-hoepfinger-hr-agrartechnik, 2023-02-08:
pls take this over also to 
Launch Terminal:
[image]

f-hoepfinger-hr-agrartechnik, 2023-02-08:
Thanks, working !!!!

closed 2023-02-08 by f-hoepfinger-hr-agrartechnik