espressif/idf-eclipse-plugin #684: No "New launch Target" for Eclipse 2022-09

Samuel-Zhang-Rheem, 2023-01-29:
**Describe the bug**
Eclipse 2022-09  without configuration bar.
After generating a new ESP project, no "sdkconfig" file, no "configuration"bar, I can't setup the target type.

**To Reproduce**

**Expected behavior**

**Screenshots**

[image]

**Espressif-IDE Product Information:**
[Espressif Product Information.txt](https://github.com/espressif/idf-eclipse-plugin/files/10529412/Espressif.Product.Information.txt)

**Eclipse Error log:** 
Window > Show View > Other > Search for "Error Log" (Attach as a file)
[.log](https://github.com/espressif/idf-eclipse-plugin/files/10529414/default.log)

sigmaaa, 2023-01-29:
Hi @Samuel-Zhang-Rheem, 
Thank you for reporting this. sdkconfig file will be generated after the build. 
Can you check, please, if there is an option in the eclipse preference to enable the launch bar: 
[image]

Samuel-Zhang-Rheem, 2023-01-29:
Hello,

The PC with Eclipse 2022 has this problem.
After tick "Enable the launch Bar", I can view the configuration, problem fixed.

Another PC with Eclipse 2021 without this issue.

Is it causing by Eclipse version 2022 or something wrong with my installation?

Regards
Sam

sigmaaa, 2023-01-29:
Hi @Samuel-Zhang-Rheem, 
Eclipse 2022 is not a problem. I have seen similar issues reported on Eclipse 2020 and 2021. Looks like there is a case when the launch bar is disabled by default after installation. It should normally be enabled by default after installation. I was never able to reproduce it, so it's hard to tell what is going wrong. Is it freshly installed Eclipse 2022 or it had some plugins installed?

Samuel-Zhang-Rheem, 2023-02-02:
There is a project, I can build it successfully in windows CMD line.
But in Eclipse, Terminal, run command line; when I go into " user\esp\esp-idf" directory, run "export.sh", it prompt "How do you to open this file",
something wrong with "Eclipse settings"?

kolipakakondal, 2023-02-02:
> export.sh

Use export.bat as mentioned here https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/get-started/windows-setup-update.html#add-esp-idf-tools-to-path-using-an-export-script

However, when you're working in Eclipse environment, always run "Espressif > ESP-IDF Install Tools > Install Tools" this will install missing tools and run export command to configure the required paths in the CDT build environment

Samuel-Zhang-Rheem, 2023-02-06:
export.bat    works.
But there is another problem:
After I update a plugin for python editor, I can't build project which is was built successfully before.
The report error message:

_Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'station'.
'int com.espressif.idf.core.build.IDFBuildConfiguration.watchProcess(java.lang.Process, org.eclipse.cdt.core.resources.IConsole)'_

How to fix it?

whoiswentz, 2023-02-12:
> export.bat works. But there is another problem: After I update a plugin for python editor, I can't build project which is was built successfully before. The report error message:
>
> _Errors occurred [...]

I'm getting this error, someone knows how to fix? It's impossible to use this plugin due this error

kolipakakondal, 2023-02-13:
Hi @Samuel-Zhang-Rheem 

Could you our latest Espressif-IDE 2.8.1 Windows Installer and see if that helps?
https://github.com/espressif/idf-installer/releases/download/espressif-ide-2.8.1-esp-idf-5.0/espressif-ide-setup-2.8.1-with-esp-idf-5.0.exe

kolipakakondal, 2023-04-13:
>Errors occurred during the build.
Errors running builder 'CDT Core Builder' on project 'station'.
'int com.espressif.idf.core.build.IDFBuildConfiguration.watchProcess(java.lang.Process, org.eclipse.cdt.core.resources.IConsole)'

This issue is fixed with IEP [v2.9.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1) release, hence closing the issue.

closed 2023-04-13 by kolipakakondal