espressif/idf-eclipse-plugin #694: Plugin's Tools Installation Wizard (Preview) doesn't work properly

JBHunterI, 2023-02-08:
**Describe the bug**
Several things don't work in the Espressif -> ESP-IDF Tools Manager -> Tools Installation Wizard (Preview). After using it to update the Plugin with the information from an ESP-IDF v5.0 installation performed with the Universal Online Installer v2.18 for Windows. There are no other errors I can see in the output from  Espressif -> ESP-IDF Tools Manager -> List installed tools, nor that of Espressif -> Product Information, yet it insists on adding entries to the PATH environment variable in Eclipse that start with **%USERPROFILE%\\.espressif\\**... when all the other environment variable it created (GIT_PATH, IDF_PATH, PYTHON_EXE_PATH), the IDF_TOOLS_PATH in Windows environment variable, start with **C:\\App\\ESP\\IDF\\**...  Also, the Tools Installation Wizard's Select Recommended button doesn't seem to work properly (see attached screenshot).  I don't know if it's related but a new 'esp32c3' IDF Target cannot be created because only 'esp' is in the drop-down list, even though my ESP32-C3-DEVKITM-1's COM port is detected (see attached screenshot). 

**To Reproduce**
Steps to reproduce the behavior:
1. Use the Universal Online Installer v2.18 for Windows to install ESP-IDF v5.0.
2. Open Eclipse, with the ESP-IDF Plugin for Eclipse already installed, selecting a new workspace.
3. Only run Espressif -> ESP-IDF Tools Manager -> List installed tools and  Espressif -> ESP-IDF Tools Manager -> Tools Installation Wizard (Preview) since ESP-IDF is already installed.
4. After each action in the previous step, keep monitoring the additions of variables in Window -> C/C++ -> Preferences ->Build -> Environment. Notice everything listed in the bug description above, an in the attached screenshots below.

**Expected behavior**
All relevant information about the properly functioning ESP-IDF installation should be gleaned from previously installation process performed with the Universal Online Installer v2.18 for Windows. No new folder **%USERPROFILE%\\.espressif\\**... should created and no paths starting with this made-up folder should be added to the newly added variables in Window -> C/C++ -> Preferences ->Build -> Environment.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
[log.txt](https://github.com/espressif/idf-eclipse-plugin/files/10680957/log.txt)

JBHunterI, 2023-02-08:
There's an entry near the top of the Eclipse Error Log from com.espressif.idf.core that says:
  `esp-idf.json file doesnt exist at this location: /C:/App/eclipseESP-IDF/`
when the only filename that looks like that under my ESP-IDF installation is: 
   `%IDF_TOOLS_PATH%\esp_idf.json`
**That's with an '_' (underscore) and not a '-' (dash)**  which BTW contains:

`{
    "$schema": "http://json-schema.org/schema#",
    "$id": "http://dl.espressif.com/dl/schemas/esp_idf",
    "_comment": "Configuration file for ESP-IDF Eclipse plugin.",
    "_warning": "Use / or \\ when specifying path. Single backslash is not allowed by JSON format.",
    "gitPath": "C:/App/ESP/IDF/tools/idf-git/2.34.2/cmd/git.exe",
    "idfToolsPath": "C:\\App\\ESP\\IDF",
    "idfSelectedId": "esp-idf-0a53fa4feea2251161728fa46bce5f8b",
    "idfInstalled": {
        "esp-idf-0a53fa4feea2251161728fa46bce5f8b": {
            "version": "5.0",
            "python": "C:/App/ESP/IDF/python_env/idf5.0_py3.8_env/Scripts/python.exe",
            "path": "C:/App/ESP/IDF/esp-idf-v5.0/"
        }
    }
}`

and the environment of the CMD shell I use to launch Eclipse from contains:
`C:\App\ESP\IDF
> set IDF && set OPEN & set ESP
IDF_CCACHE_ENABLE=1
IDF_GIT=C:/App/ESP/IDF/tools/idf-git/2.34.2/cmd/git.exe
IDF_GIT_DIR=C:\App\ESP\IDF\tools\idf-git\2.34.2\cmd\
IDF_PATH=C:\App\ESP\IDF\esp-idf-v5.0
IDF_PYTHON=C:/App/ESP/IDF/python_env/idf5.0_py3.8_env/Scripts/python.exe
IDF_PYTHON_DIR=C:\App\ESP\IDF\python_env\idf5.0_py3.8_env\Scripts\
IDF_PYTHON_ENV_PATH=C:\App\ESP\IDF\python_env\idf5.0_py3.8_env
IDF_TARGET=esp32c3
IDF_TOOLS_PATH=C:\App\ESP\IDF
OPENOCD_SCRIPTS=C:\App\ESP\IDF\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts
ESPBAUD=115200
ESPPORT=COM8
ESP_IDF_VERSION=5.0
`
I don't know what else the Plugin needs to know.
Why would the Plugin looking for a) the wrong filename, and b) anywhere other than in the %IDF_TOOLS_PATH% folder?

JBHunterI, 2023-02-09:
FYI, I just installed the exact same ESP-IDF v5.0 from the Universal Online Installer v2.18 into a new %USERPROFILE%\\.espressif folder and launched the exact same Eclipse from the ESP-IDF CMD shell. Still don't see esp32c3 as an available target, but the Tools Installation Wizard shows everything is installed with green light next to each item. The Error Log still shows this "esp-idf.json file doesnt exist at this location: /C:/App/eclipseESP-IDF/" message from com.espressif.idf.core also.

kolipakakondal, 2023-02-09:
Hi @alirana01 Could you check this?

JBHunterI, 2023-02-10:
I uninstalled the ESP-IDF v5.0 and installed  ESP-IDF v4.4.4 from the Universal Online Installer v2.20 into a new %USERPROFILE%\.espressif folder and copied (xcopy /s/i) the blink example into a new workspace from the ESP-IDF 4.4 CMD shell, built/ flashed/monitored it onto my ESP32-C3-DevKitM-1, as a 'esp32c3' target, using the idf.py command: everything worked fine. I launch Eclipse from the same ESP-IDF 4.4 CMD shell, but no matter what I do, I cannot create a ESP32-C3 target in the Eclipse Launch Bar after running Espressif -> ESP-IDF Tools Manager -> Tools Installation Wizard (Preview) to install only the tools appropriate for the ESP32-C3 target. The  'esp32c3' target just isn't available from the Target drop-down list, only the  'esp' target is.  Why is that?

The only Eclipse environment variable that creates are:
```
GIT_PATH = %USERPROFILE%\.espressif\tools\idf-git\2.34.2\cmd\git.exe
IDF_COMPONENT_MANAGER = 1
IDF_PATH = %USERPROFILE%\.espressif\esp-idf-v4.4.4
PYTHON_EXE_PATH = %USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
```
Notice there's no IDF_MAINTAINER, OPENOCD_SCRIPTS  or PATH variables created with ESP-IDF-specific additions.

I tried upgrading all the Eclipse software, including the Espressif module, the Product Information looks fine:
Executing %USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.EXE %USERPROFILE%\.espressif\esp-idf-v4.4.4\tools\idf_tools.py list
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 11.2_20220823 (recommended)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 11.2_20220823 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-2021r2-patch5-8.4.0 (recommended)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-2021r2-patch5-8.4.0 (recommended)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-2021r2-patch5-8.4.0 (recommended)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-2021r2-patch5-8.4.0 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended)
* cmake: CMake build system
  - 3.23.1 (recommended, installed)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20221026 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.3 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.9 (recommended)

CDT Build environment variables
PATH: %USERPROFILE%\.espressif\tools\idf-git\2.34.2\cmd;/App/eclipseESP-IDF//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.4.v20220903-1038/jre/bin/server;/App/eclipseESP-IDF//plugins/org.eclipse.justj.openjdk.hotspot.jre.full.win32.x86_64_17.0.4.v20220903-1038/jre/bin;%USERPROFILE%\.espressif\tools\riscv32-esp-elf-gdb\11.2_20220823\riscv32-esp-elf-gdb\bin;%USERPROFILE%\.espressif\tools\riscv32-esp-elf\esp-2021r2-patch5-8.4.0\riscv32-esp-elf\bin;%USERPROFILE%\.espressif\tools\cmake\3.23.1\bin;%USERPROFILE%\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\bin;%USERPROFILE%\.espressif\tools\ninja\1.10.2\;%USERPROFILE%\.espressif\tools\idf-exe\1.0.3\;%USERPROFILE%\.espressif\tools\ccache\4.3\ccache-4.3-windows-64;%USERPROFILE%\.espressif\esp-idf-v4.4.4\tools;%USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env\Scripts\;%USERPROFILE%\.espressif\tools\idf-git\2.34.2\cmd\;%USERPROFILE%\.espressif;C:\Program Files\Common Files\Oracle\Java\javapath;...my existing Windows paths...;;%USERPROFILE%\ESP-IDF_ws\blink
...
OPENOCD_SCRIPTS: %USERPROFILE%\.espressif\tools\openocd-esp32\v0.11.0-esp32-20221026\openocd-esp32\share\openocd\scripts
IDF_CCACHE_ENABLE: 1
IDF_PYTHON_DIR:  %USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env\Scripts\
IDF_GIT_DIR:  %USERPROFILE%\.espressif\tools\idf-git\2.34.2\cmd\
TEMP_IDF_PYTHON_PATH: "%USERPROFILESHORT%\AppData\Local\Temp\idf-python-path.txt"
IDF_PYTHON: %USERPROFILE%/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe
IDF_COMPONENT_MANAGER: 1
PREFIX: %USERPROFILE%/.espressif/python_env/idf4.4_py3.8_env/Scripts/python.exe  %USERPROFILE%/.espressif/esp-idf-v4.4.4/
PYTHON_EXE_PATH:  %USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env\Scripts\python.exe
GIT_PATH:  %USERPROFILE%\.espressif\tools\idf-git\2.34.2\cmd\git.exe
ESPBAUD: 115200
IDF_GIT: %USERPROFILE%/.espressif/tools/idf-git/2.34.2/cmd/git.exe
IDF_TOOLS_PATH:  %USERPROFILE%\.espressif
=C:: %USERPROFILE%\ESP-IDF_ws\blink
TEMP_IDF_PATH: %USERPROFILESHORT%\AppData\Local\Temp\idf-path.txt
ESPPORT: COM8
PYTHONNOUSERSITE: True
IDF_PATH:  %USERPROFILE%\.espressif\esp-idf-v4.4.4
PARAM: esp-idf-82c7bfec80c3f6504e8d7772f06b6945
IDF_TARGET: esp32c3
IDF_PYTHON_ENV_PATH: %USERPROFILE%\.espressif\python_env\idf4.4_py3.8_env
TEMP_IDF_GIT_PATH: "%USERPROFILESHORT%\AppData\Local\Temp\idf-git-path.txt"

Operating System: windows 10
Java Runtime Version: 17.0.4.1+1
Eclipse Version: 4.26.0.v20221123-1800
Eclipse CDT Version: 11.0.0.202211062329
IDF Eclipse Plugin Version: 2.8.1.202302071030     <---- upgraded from v2.7 & all Eclipse software via Help -> Check for Updates
ESP-IDF v4.4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.8.7

In fact, when I import the blink IDF project to see it in the Eclipse Project Explorer, I see an error pop-up saying the target was changed from 'esp32c3' to 'esp'.  Why?!?!?
[image]

JBHunterI, 2023-02-11:
I uninstalled the USB JTAG & Serial drivers in Device Manager, then uninstalled the previous ESP-IDF v4.4.4 installation, rebooted my PC, made a full install of **all** tools & for **all** targets in %USERPROFILE%\\.espressif, added the blink example, and I still get: 
[image]
What the heck is an "esp" target, and where's my "esp32c3" target in the Plugin when my %USERPROFILE%\\.espressif\\idf-env.json says:
`{
    "idfInstalled": {
        "sha": {
            "targets": [
                "esp32",
                "esp32s2",
                "esp32s3",
                "esp32c3"
            ]
        }
    }
}`
There's no "esp" target listed in there, but there is an "esp32c3" target.  Am I the anyone seeing this problem?
In fact, if I follow [Select the Target Chip: set-target](https://docs.espressif.com/projects/esp-idf/en/v5.0/esp32c3/api-guides/tools/idf-py.html#select-the-target-chip-set-target) and do:
%USERPROFILE%\\ESP-IDF_ws\\blink> **idf.py --list-targets**
esp32
esp32s2
**esp32c3            <--- there it is!!!**
esp32s3

I've never had to before, but at this point I'll try anything:
%USERPROFILE%\\ESP-IDF_ws\\blink> **idf.py set-target esp32c3**
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
**Set Target to: esp32c3**, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
. . .

It never has before, but could the Plugin be having problems with the fact that my ESP32-C3-DevKitM-1 is using a USB Serial/JTAG connection to my PC, instead of its default separate UART0 and JTAG ports? Before I even get to change Component config → ESP System Settings → "Channel for console output" from (Default: UART0) to (USB Serial/JTAG Controller), the messages shown as 'idf.py menuconfig' starts-up seem to show that the ESP32-C3 is already the detected target:

%USERPROFILE%\\ESP-IDF_ws\\blink> **idf.py menuconfig**
Executing action: menuconfig
. . .
-- Building ESP-IDF components for **target esp32c3**

but after I change it & re-enter Eclipse, I get this pop-up:
[image]

May I please have some sort of response to all/any of this?

JBHunterI, 2023-02-14:
Starting Eclipse **not** from the ESP-IDF CMD shell (where **everything is already defined** messes up the Plugin, I guess?!?!?) and **manually** adding IDF_TOOLS_PATH to the empty Eclipse environment (don't know why that would help since it's already in the Windows environment) before running Espressif -> ESP-IDF Tools Manager -> Tools Installation Wizard (Preview) seems to fill the Eclipse environment with all 9+ expected variables (including IDF_MAINTAINER, OPENOCD_SCRIPTS and PATH) by re-installing **everything the ESP-IDF Universal Installer had already installed** to make the ESP-IDF CMD shell functional. Then, a "esp32c3" target is available in Eclipse, **even with _only_ ESP32-Cx RISC-V targets installed somewhere _other than_ %USERPROFILE%\\.espressif**.

alirana01, 2023-05-24:
@JBHunterI can you please try the installation wizard with the latest release. You can reopen the issue if the problem persists.

This was resolve with recent releases

closed 2023-05-24 by alirana01