espressif/idf-eclipse-plugin #696: Esp32 s3 usb-jtag debug error (uxTopUsedPriority)

ertekinsener, 2023-02-08:
hardware:esp32s3 devkit C1
software: espressif-ide-setup-2.8.0-with-esp-idf-5.0 (latest)
when I try to debug I get this error and I can't debug.
If I delete my program with Revo Uninstaller and reinstall it, it works as expected on the first run.

**Screenshots**
[image]
[image]
[image]
[image]

**Espressif-IDE Product Information:**
NUMBER_OF_PROCESSORS: 16
IDF_PYTHON_ENV_PATH: C:\Espressif\python_env\idf5.0_py3.8_env

Operating System: windows 11
Java Runtime Version: 11.0.17+8-LTS
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.8.0.202301311144
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
WARNING: Git was not found
error: cannot spawn git: No such file or directory
fatal: Could not run 'git status --porcelain=2' in submodule vendor/c_exception
fatal: 'git status --porcelain=2' failed in submodule components/cmock/CMock
WARNING: Git version unavailable, reading from source
ESP-IDF v5.0.0
Python set for IDF_PYTHON_ENV: Python 3.8.7

**Eclipse Error log:** 
there are lots off error logs, 
[image]
[image]

Console output
Open On-Chip Debugger v0.11.0-esp32-20221026 (2022-10-26-14:48)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Flashing C:/Espressif/frameworks/esp-idf-v5.0/workspace/geegqwagq/build/bootloader/bootloader.bin at 0x0
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : esp_usb_jtag: serial (F4:12:FA:FA:00:C4)
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
Warn : target esp32s3.cpu1 examination failed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Target halted, PC=0x40041A79, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (1) - (Power on reset)
Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
Info : Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
[... 150 lines cut ...]
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Error: FreeRTOS: uxTopUsedPriority is not defined, consult the OpenOCD manual for a work-around!
Error: FreeRTOS: uxTopUsedPriority is not defined, consult the OpenOCD manual for a work-around!
Info : JTAG tap: esp32s3.cpu0 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : JTAG tap: esp32s3.cpu1 tap/device found: 0x120034e5 (mfg: 0x272 (Tensilica), part: 0x2003, ver: 0x1)
Info : [esp32s3.cpu0] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Debug controller was reset.
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x500000EF, debug_reason=00000000
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu1] requesting target halt and executing a soft reset
Info : [esp32s3.cpu0] Core was reset.
Info : [esp32s3.cpu0] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Debug controller was reset.
Info : [esp32s3.cpu1] Core was reset.
Info : [esp32s3.cpu1] Target halted, PC=0x40000400, debug_reason=00000000
Info : [esp32s3.cpu1] Reset cause (3) - (Software core reset)
Info : [esp32s3.cpu0] Reset cause (3) - (Software core reset)

kolipakakondal, 2023-02-14:
> Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000

This line might indicate that the app is configured in single-core mode, so the 2nd CPU is powered down. Please check.

ertekinsener, 2023-02-17:
> > Error: [esp32s3.cpu1] Unexpected OCD_ID = 00000000
> 
> This line might indicate that the app is configured in single-core mode, so the 2nd CPU is powered down. Please check.

If it is something that can be changed from menuconfig, I did not change it. I got the same error in esp idf examples. Trying again with esp ide 2.7 (esp idf 5.0) fixed the problem

kolipakakondal, 2023-02-20:
Thanks @ertekinsener for updating. 

> IEP-831: Set remotetimeout to 20 sec for GDB Client in https://github.com/espressif/idf-eclipse-plugin/pull/676

This was the only change we made in IDE 2.8.0 in the OpenOCD debugging config. This would not have impacted the reported error.

Anyway, I will keep this ticket open for some more time, If you come across the same problem please report us for troubleshooting further.

kolipakakondal, 2023-04-13:
Let us know if there is anything that we could help on this. Closing this issue for now, feel free to open it if the issue still persists.

closed 2023-04-13 by kolipakakondal