espressif/idf-eclipse-plugin #697: pls take this over also to Launch Terminal:

f-hoepfinger-hr-agrartechnik, 2023-02-08:
pls take this over also to 
Launch Terminal:
[image]

_Originally posted by @f-hoepfinger-hr-agrartechnik in https://github.com/espressif/idf-eclipse-plugin/issues/683#issuecomment-1422896116_

kolipakakondal, 2023-02-09:
Hi @f-hoepfinger-hr-agrartechnik Could you please add more details about the bug? What is the problem and what is the expected behavior?

f-hoepfinger-hr-agrartechnik, 2023-02-11:
ESP-IDF Serial Monitor makes only sense on ESP-IDF Projects. so other Projects should not be listed there. 

you made this already for the Top dropdown list.

kolipakakondal, 2023-02-13:
Ok, thanks for the feedback

kolipakakondal, 2023-02-13:
https://github.com/espressif/idf-eclipse-plugin/pull/699

kolipakakondal, 2023-03-29:
This is fixed as part of v2.9.0 release, please check

closed 2023-03-29 by kolipakakondal