espressif/idf-eclipse-plugin #701: Documentation:  idf-eclipse-plugin/README.md does not include instructions for launch bar

cjearls-witekio, 2023-02-13:
Issue: https://github.com/espressif/idf-eclipse-plugin#readme 
Under the header "Configuring Launch target" the instructions are incredibly vague and do not guide the user to how to enable the launch bar if it is not properly enabled. Text in question follows below.

Next, we need to tell CDT to use the toolchain for our project so that all the headers will be indexed and resolved. This is accomplished through the Launch Bar, the new widget set you see on the far left of the toolbar. 

_Maybe include an  image of the gui with the launch bar enabled here_
[image]

**Note:** This will be shown only when you have a project in the project explorer. **If the launch bar is not displayed it can be enabled by going to ```Wndow ``` > ```Preferences``` Then type in the Filter at the top left "Launch Bar" Click on Launch bar from the menu, then tick ```Enable the Launch Bar.``` This allows the launch bar to be displayed if it was not previously.**
[image]

This way it is a bit more descriptive, as the launch bar may or may not always be present depending on a user's configuration.

kolipakakondal, 2023-02-14:
Thanks for sharing the feedback @cjearls-witekio We will improve this

kolipakakondal, 2023-03-29:
Added to the FAQ doc
https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-enable-the-launchbar-in-espressif-ide

closed 2023-03-29 by kolipakakondal