espressif/idf-eclipse-plugin #710: Doesn't show ESP32C6 in ESP Target list

RJSDevel, 2023-02-28:
**Describe the bug**
Doesn't show ESP32C6 in ESP Target list, doens't constaint any preview chips

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Espressif-IDE
2. Open Zigbee Example
3. Click to ESP targets list 
4. No ESP32C6

**Expected behavior**
I've edited idf_actions.py to show all chips without any dialogs, and in idf --list-tagets whows all targets then, but plugin does'nt 

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
* xtensa-esp-elf-gdb: GDB for Xtensa
  - 12.1_20221002 (recommended, installed)
* riscv32-esp-elf-gdb: GDB for RISC-V
  - 12.1_20221002 (recommended, installed)
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp-clang: Toolchain for all Espressif chips based on clang (optional)
  - 15.0.0-23786128ae (recommended, installed)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
  - esp-12.2.0_20230208 (recommended, installed)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
  - 2.35_20220830 (recommended, installed)
* cmake: CMake build system
  - 3.24.0 (recommended, installed)
  - 3.16.3 (supported)
* openocd-esp32: OpenOCD for ESP32
  - v0.11.0-esp32-20221026 (recommended, installed)
* ninja: Ninja build system
  - 1.10.2 (recommended, installed)
* idf-exe: IDF wrapper tool for Windows
  - 1.0.3 (recommended, installed)
* ccache: Ccache (compiler cache)
  - 4.6.2 (recommended, installed)
* dfu-util: dfu-util (Device Firmware Upgrade Utilities)
  - 0.11 (recommended, installed)
* esp-rom-elfs: ESP ROM ELFs
  - 20230113 (recommended, installed)
IDF_PYTHON_ENV_PATH: D:\Espressif\python_env\idf5.1_py3.7_env
ESP_ROM_ELF_DIR: D:\Espressif\tools\esp-rom-elfs\20230113\
Operating System: windows 10
Java Runtime Version: 17.0.6+10-LTS
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.8.1.202302071030
ESP-IDF v5.1-dev-3663-g8486a2c3ad-dirty
Python set for IDF_PYTHON_ENV: Python 3.7.9

**Eclipse Error log:** 
[log.log](https://github.com/espressif/idf-eclipse-plugin/files/10851341/log.log)

!ENTRY com.espressif.idf.core 4 0 2023-02-28 19:23:58.417
!MESSAGE Cannot run program "D:/Espressif/frameworks/esp-idf-master//tools/idf_tools.py" (in directory "\"): CreateProcess error=193, %1 не является приложением Win32
!STACK 0

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

RJSDevel, 2023-02-28:
May be related https://github.com/espressif/idf-eclipse-plugin/commit/2a69ff6813794c77d0ff5e91136a075a704e3a6e

RJSDevel, 2023-02-28:
The matter I can't add in New Launch Target too, it has't got esp32c6 too.

RJSDevel, 2023-02-28:
[image]

kolipakakondal, 2023-02-28:
> [image]

1. Add Name - esp32c6
2. In the target list, replace esp32 with the esp32c6. This will be added as a new entry.
3. Once you close the dialog, you will be able to see esp32c6 in the IDE toolbar target list.

However, I just ran a quick test and realized ESP-IDF 5.1-dev is not update-to-date with the master and doesn't have esp32c6 support. Hence, I would recommend using the latest branch of esp-idf for esp32c6 support for now.

After moving to master, you need to do a couple of other additional settings to start using the esp32c6 toolchain (once we start supporting esp32c6 in IDE, all the below steps will be automated). For now, here you go - 
1. Move to esp-idf master
2. Create esp32c6 launch target manually https://github.com/espressif/idf-eclipse-plugin#configuring-launch-target
3. Configure 'Core Build Toolchains' for esp32c6 as mentioned [here](https://github.com/espressif/idf-eclipse-plugin#configuring-core-build-toolchains) 
4. Configure CMake Toolchain for esp32c6 as mentioned [here](https://github.com/espressif/idf-eclipse-plugin#configuring-cmake-toolchain) 
5. Then you're good to build with the esp32c6 target

Maybe I could try to capture a screen recording of the above steps a bit later if required.

RJSDevel, 2023-03-01:
> esp-idf v5.0 master branch [image]
> 
> esp-idf v5.1-dev branch (I can't find esp32c6 here) [image]

I guess that because we need use --preview key to get new chips

RJSDevel, 2023-03-01:
> esp-idf v5.0 master branch [image]
> 
> esp-idf v5.1-dev branch (I can't find esp32c6 here) [image]

How should I get master? Either just clone from master branch or something else way?

kolipakakondal, 2023-03-01:
Hello,

You could choose 'master' from the list and install it.
https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf

[image]

RJSDevel, 2023-03-01:
> Hello,
> 
> You could choose 'master' from the list and install it. https://github.com/espressif/idf-eclipse-plugin#installing-esp-idf
> 
> [image]

that's exactly what I did, I don't know why the installer clone the dev

RJSDevel, 2023-03-01:
> 1. Add Name - esp32c6
> 2. In the target list, replace esp32 with the esp32c6. This will be added as a new entry.
> 3. Once you close the dialog, you will be able to see esp32c6 in the IDE toolbar [...]

unfortunately no any available toolchains

[image]

kolipakakondal, 2023-06-13:
Now we have added esp32-c6 with the IEP v2.10.0 release. Please take a look https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.10.0

closed 2023-06-13 by kolipakakondal