espressif/idf-eclipse-plugin #719: ESP32 Eclipse debugging with Segger Jlink

Samuel-Zhang-Rheem, 2023-03-22:
I'm using Segger Jlink for several project with different MCU, including ESP32.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/jtag-debugging/configure-ft2232h-jtag.html?highlight=configure%20usb%20driver
describe Zadig tool to setup USB driver,  should I change the Jlink driver to WinUSB driver as well for ESP32 MCU debugging?
[image]

When I working for other MCU, how can I return the Segger debugger to Jlink driver instead of WinUSB driver?

erhankur, 2023-03-24:
@Samuel-Zhang-Rheem I tried on windows with a Jlink-EDU. It is necessary to install WinUSB(libUSB) with Zadig to get OpenOCD working. And in the same way, you can return to the original jlink driver for other setups.

Samuel-Zhang-Rheem, 2023-03-26:
1. How to return the Jlink-EDU back to its original driver (Jlink driver) from WinUSB driver? 
2. To debug ESP32 MCU, Is it possible to do "GDB Hardware Debugging" with Jlink  debugger at Jlink driver? How to setup the parameters in the configuration?  Jlink is much faster than OpenOCD.

[image]

erhankur, 2023-03-26:
> How to return the Jlink-EDU back to its original driver (Jlink driver) from WinUSB driver?

You can uninstall WinUSB driver from Device Manager until you see jlink in the `Universal Serial Bus Controllers` section. just be sure to select the `Attempt to remove the driver for this device` box in each uninstall. 

> To debug ESP32 MCU, Is it possible to do "GDB Hardware Debugging" with Jlink debugger at Jlink driver? How to setup the parameters in the configuration? Jlink is much faster than OpenOCD.

SEGGER Jlink software doesn't support Espressif devices yet. So you will need to setup OpenOCD

Samuel-Zhang-Rheem, 2023-04-04:
SEGGER Jlink software support many microcontrollers, the reason doesn't support is that SEGGER is not add the interface for ESP32 or Espressif need add something in the core?

I'm design WiFi with safety critical (Class B) control, we are using a small microcontroller to do the safety function and  ESP32 do the WiFi.  ESP32 is SoC with A/D and digital output, that is enough for my application, if ESP32 has safety critical library, the ESP32 will be the only MCU on my board, that will save cost.

Is ESP32 has safety critical library?

igrr, 2023-04-13:
> SEGGER Jlink software support many microcontrollers, the reason doesn't support is that SEGGER is not add the interface for ESP32 or Espressif need add something in the core?

Jlink software does not automatically support the ESP32 because some aspects of ESP32 are different from other MCUs. (For example, each Xtensa core has a slightly different configuration, which the debugger needs to know about. Each MCU has a slightly different way of writing the program or breakpoints to Flash. The memory map is also different.) All these things need to be implemented in Jlink software, and this hasn't been done. I will cross-reference Segger's reply on this topic: https://forum.segger.com/index.php/Thread/8470-SOLVED-ESP32-target-support/

Regarding the critical safety library, this is something we are planning to support, although it's not yet decided  which chip families will be supported (e.g. ESP32-S3 and newer, or ESP32 as well). Could you please email sales at espressif.com and reference this ticket? We can continue over email.

hardphoenix, 2023-10-12:
Hi 
You can use Example Of ESP32 Jllink In This Repo:

[](https://github.com/hardphoenix/ESP32-JLink)