espressif/idf-eclipse-plugin #731: Workspace hangs when closing and opening projects...

davidlehrian, 2023-04-01:
**Describe the bug**
I created a brand new empty workspace and I created 3 esp-idf example projects, blink, hello_world and generic_gpio. I built the workspace. I close 2 of the projects and then attempt to open them and it hangs on "Refreshing workspace: (94%). I have to kill Eclipse. Sometimes I can open them once or twice, but it eventually hangs. 

**Expected behavior**
I expect to be able to close and open a project without having to kill Eclipse. 

**Screenshots**
[image]

**Espressif-IDE Product Information:**
[Espressif.txt](https://github.com/espressif/idf-eclipse-plugin/files/11130182/Espressif.txt)

**Eclipse Error log:** 
[error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/11130196/error_log.log)

Please attach the error log as described here https://github.com/espressif/idf-eclipse-plugin#error-log

sigmaaa, 2023-04-03:
Hi @davidlehrian,

Thank you for reporting this issue. Created a PR, that should fix it

closed 2023-04-04 by kolipakakondal

davidlehrian, 2023-04-04:
I built the latest master code and in my test case and production environment, which has a dozen projects, it is working. I can open and close projects at will. Thank you for the quick response.