espressif/idf-eclipse-plugin #734: Building Workspace stops and never finishes building... v2.8.1 on 2022-09 Eclipse

Coffeeye, 2023-04-03:
**Describe the bug**
Since few days Im discovering a huge issue with Eclipse and Espressif env becouse of never finishing process of building workspace. 
I have tried this on multiple stations or virtualenv with the same resoult. The installations were fresh or not, which results in the same. Example compiles and everything works just fine but after few include's building process just hang with no error and it looks like it never finishes, I even gave my self a half hour but it didnt move even by 1% (but the animation says its working...)

building with console 'idf.py build' passes on each/any project and env tested (mentioned above)

**To Reproduce**
I would say just start a new project but it wont happen until you start modify the project the common way.

**Expected behavior**
Project build.

**Screenshots**
[image]

**Espressif-IDE Product Information:**
[product_info.log](https://github.com/espressif/idf-eclipse-plugin/files/11137391/product_info.log)

**Eclipse Error log:** 
[log.log](https://github.com/espressif/idf-eclipse-plugin/files/11137405/log.log)

ammaree, 2023-04-03:
+1 same problem since I installed the latest. Also remove complete Eclipse 2022-09 installation, reinstalled from scratch, same problem.

Also tried the same process with Eclipse 2022-12 but no better so reverted to 2022-09 and simply doing the build outside of Eclipse in a Terminal session.

Hangup only occurs if a warning or error occurs during the build process.

Coffeeye, 2023-04-03:
problems just waterfall on me, cant open sdkconfig anymore and eclipse alerts me about it cant find binnaries anymore...

Coffeeye, 2023-04-03:
"Downgrading" plugin to (i choosed) [2.4.0](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.4.0) seems to fix the issue (atleast to the level its compiling)

Steps : 
> Uninstall current plugin from Eclipse Marketplace -> Installed
> Download the archive containing the plugin.
> Open Eclipse and go to the "Help" menu.
> Select the "Install New Software" option.
> Click the "Add" button to add a new software source.
> Enter the name and location of the plugin you want to install.
> Click the "OK" button to add a new software source.
> Select a new software source from the list and select an older plug-in version.
> Click "Next" and then accept the license terms.
> Click "Finish" to start the installation process.

After the installation is complete, restart Eclipse for the changes to take effect.

kolipakakondal, 2023-04-03:
Hi All, Sorry about this issue! We will be releasing the fix tomorrow.

Before that, I need your help in confirming this fix.

Here is our build [udpatesite](https://github.com/espressif/idf-eclipse-plugin/suites/11940104753/artifacts/626365158) zip file for the fix 

Here are the [instructions](https://github.com/espressif/idf-eclipse-plugin#installing-idf-eclipse-plugin-from-local-archive) on how to install them 

Thanks

ammaree, 2023-04-03:
@kolipakakondal 

Did not work for me
`An error occurred while collecting items to be installed
session context was:(profile=_Users_andremaree_eclipse_cpp-2022-09_Eclipse.app_Contents_Eclipse, phase=org.eclipse.equinox.internal.p2.engine.phases.Collect, operand=, action=).
No repository found containing: osgi.bundle,com.espressif.idf.branding,2.9.0.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.core,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.debug.gdbjtag.openocd,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.help,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.launch.serial.core,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.launch.serial.ui,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.sdk.config.core,1.0.2.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.sdk.config.ui,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.serial.monitor,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.terminal.connector,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.terminal.connector.serial,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.ui,1.0.1.202303310948
No repository found containing: osgi.bundle,com.espressif.idf.wokwi,1.0.0.202303310948
No repository found containing: org.eclipse.update.feature,com.espressif.idf.feature,2.9.0.202303310948`

kolipakakondal, 2023-04-03:
> @kolipakakondal
>
> Did not work for me `An error occurred while collecting items to be installed session context was:(profile=_Users_andremaree_eclipse_cpp-2022-09_Eclipse.app_Contents_Eclipse, [...]

Hi @ammaree  This could be due to the downloaded update site content is corrupted for some reason. Can you download and install again? I just did and it seems to be working fine.

Maksons, 2023-04-03:
Linux , IDE 2.9.0  problem simillar :
After Build OK , could not close Espreessif IDE (closing in progres forever)...

Tried given patch - lost Espressif build functionality of Eclipse(Espressif IDE) at all :( 
No build (only blink.elf) , no way to install toolst/etc...

vtzann, 2023-04-03:
I have the same issue, it gets stuck if it comes across a warning. In my case, in a switch statement case is allowed to fall-through, generating a compiler warning That causes it to get stuck. Placing a goto statement fixes it.
Initially it happened on ESP-IDE. I also installed the plugin on eclipse 2022-09 with the same results.
Applying the update as suggested, did not fix the problem in either case.

kolipakakondal, 2023-04-06:
Hi All,

We have released v2.9.1 with this fix, please get the latest.

Installing IDF Plugins using the Update site URL:
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

Installing IDF Plugins using the local archive file:
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-2.9.1.zip

Espressif IDE
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-win32.win32.x86_64.zip
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-linux.gtk.x86_64.tar.gz

vtzann, 2023-04-06:
So, far so good! It appears to be working.
Thank you very much for the fix.

 

https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-2.9.1.zip
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-win32.win32.x86_64.zip
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-linux.gtk.x86_64.tar.gz

kolipakakondal, 2023-04-12:
As this is fixed with [v2.9.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1), closing this issue.

closed 2023-04-12 by kolipakakondal