espressif/idf-eclipse-plugin #738: java.io.FileNotFoundException compile_commands.json

RJSDevel, 2023-04-04:
**Describe the bug**
Can't build the project

**To Reproduce**
Steps to reproduce the behavior:
1. Create new Zigbee project
2. Try to built it
3. Get an error
4. See error

**Expected behavior**
Built an project

**Espressif-IDE Product Information:**
[info.txt](https://github.com/espressif/idf-eclipse-plugin/files/11151251/info.txt)
**Eclipse Error log:** 
[error_log.log](https://github.com/espressif/idf-eclipse-plugin/files/11151252/error_log.log)

kolipakakondal, 2023-04-05:
Hi @RJSDevel  Sorry for the issue. Are you trying to build with a custom build directory?

RJSDevel, 2023-04-05:
Hi, no I don't, just in workspace.
I've created the project from template zigbee_light for esp32c6

sigmaaa, 2023-09-22:
Hi @RJSDevel,
thank you for reporting this. It's fixed now, verified on 2.11 but should be okay on the previous version as well.

closed 2023-09-22 by sigmaaa