espressif/idf-eclipse-plugin #743: Ide stuck building when unsed var has appeared

RJSDevel, 2023-04-08:
**Describe the bug*
Ide stuck building when unsed var has appeared

**To Reproduce**
Steps to reproduce the behavior:
Create a progect from http_server and make any unused variable and try to build it

**Expected behavior**
No any stuck

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Espressif-IDE Product Information:**
[info.txt](https://github.com/espressif/idf-eclipse-plugin/files/11183919/info.txt)

**Eclipse Error log:** 
[errors.log](https://github.com/espressif/idf-eclipse-plugin/files/11183922/errors.log)

kolipakakondal, 2023-04-09:
This is a a duplicate of https://github.com/espressif/idf-eclipse-plugin/issues/723 and fixed in the recent release of eclipse plugin v2.9.1

Please try with the latest verison and let us know how it goes.

Please find the release notes here
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1

Please use below update site to get the latest version
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

kolipakakondal, 2023-04-12:
As this is fixed with [v2.9.1](https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1), closing this issue.

closed 2023-04-12 by kolipakakondal