espressif/idf-eclipse-plugin #744: ESP-IDF Serial Monitor doesn't display "Project name:"

davidlehrian, 2023-04-08:
The Launch Terminal->ESP-IDF Serial Monitor doesn't display any "Project name:"s

**To Reproduce**
Create a new workspace.
Create a new hello_world example project.
Run the project.
Open Window->Show view->Terminal
Click on "Open a Terminal" button
Select ESP-IDF Serial Monitor
There are no "Project name:"s available to be selected. 

**Expected behavior**
Historically the open project names have been visible and available to be selected. 

**Screenshots**
I couldn't capture the empty drop down, but when I click on the drop down no projects are available to choose.
[image]

**Espressif-IDE Product Information:**
I am running the nightly build so that I have the fix for PR #729 
[Expressif Product Information.txt](https://github.com/espressif/idf-eclipse-plugin/files/11183950/Expressif.Product.Information.txt)

**Eclipse Error log:** 
[Eclipse Error Log.log](https://github.com/espressif/idf-eclipse-plugin/files/11183952/Eclipse.Error.Log.log)

kolipakakondal, 2023-04-09:
Hi @davidlehrian Please use the latest release of plugin v2.9.1 which has a fix for both - the serial monitor dialog not showing up project list and the project build UI block issue.

Please find the release notes here
https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.9.1

**Here is how to get the latest plugin**

Installing IDF Plugins using the Update site URL:
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/latest/

Installing IDF Plugins using the local archive file:
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/com.espressif.idf.update-2.9.1.zip

Espressif IDE
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-aarch64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-macosx-cocoa-x86_64-v2.9.1.dmg
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-win32.win32.x86_64.zip
https://dl.espressif.com/dl/idf-eclipse-plugin/ide/Espressif-IDE-2.9.1-linux.gtk.x86_64.tar.gz

davidlehrian, 2023-04-09:
I have updated to the latest build 2.9.1.202304081133 and it does indeed fix the issue described in this bug. Thanks for the quick fix!

closed 2023-04-10 by kolipakakondal