espressif/idf-eclipse-plugin #745: espressif-ide:couldn't open after changing the language config.

2447328758, 2023-04-09:
**Describe the bug**
I just installed the espressif-ide yesterday. Today i altered the language config, and reopen it when the bug occurs. Here are the operations I made earlier:

**To Reproduce**
1. I imoprt a project into the workspace .The project is under the [/examples/get-started/hello_world](https://github.com/espressif/esp-idf/tree/master/examples/get-started/hello_world). And change the file without saving.
2. I found that some include files is not found by the ide, so i reset the esp-idf path in the menu window.
3. then i alter the language from english to chinese in the menu bar.
4. The ide noted me to restart the it and at the same time the editor notex me to choose to save or not. And I chose not to save  and to restart the ide.
5. And then every time i open the ide, it shows an error dialog window that wish me to check the appliction log file.

**Expected behavior**
I expected to solve the problem and restart it successfully.Possibly I would like to find the reason for it.

**Screenshots**
When I restart it, the dialog will show up.
[image]
And I checked the log file. The contents are as follows:
The [log file](https://github.com/2447328758/esp-idf-ide_issue_report/blob/main/1681005993950.log) is so large that I have upload it to my github repo.
> problem.

**Espressif-IDE Product Information:**
Operating System: windows 10
Java Runtime Version: not sure, maybe 1.8 or the idf-env installed java 17
Eclipse Version: 4.25.0 Realease
IDF Eclipse Plugin Version: 2.9.0
esp-idf-v5.0.1
Python set for IDF_PYTHON_ENV: Python 3.11.2

**Eclipse Error log:** 
The [log file](https://github.com/2447328758/esp-idf-ide_issue_report/blob/main/1681005993950.log) is so large that I have upload it to my github repo. [](https://github.com/espressif/idf-eclipse-plugin#error-log 
)

2447328758, 2023-04-09:
Now I found that the bug occurs as long as i change the language. It all ablout the language. By the way, the error log file showed that it use the java 8, but the espressif-ide2.9 need java11+.So I reinstall the ide and the java, the problem  was sovled. The problem is that why after i change the language it alter the jvm path automatically. The java 8 is installed by myself whick is added to the system path variable and the java 17 is installed embeded by the ide installer. So i was trying to find the reason for the change for jvm and how to config it maybe.

kolipakakondal, 2023-04-09:
Hi @2447328758 Thanks for reporting the issue.

> By the way, the error log file showed that it use the java 8, but the espressif-ide2.9 need java11+

You got it right,  Espressif-IDE 2.8.0 higher needs Java 11 and higher

We will investigate further why changing the language is changing the Java verison

2447328758, 2023-04-11:
Yes,I am glad to do that.
before:
```
-vm
E:\Espressif\tools\amazon-corretto-11-x64-windows-jdk\jdk17.0.6_10\bin\javaw.exe
```
[image]

after:
```
-nl
zh
-vmargs
-vm
E:\Espressif\tools\amazon-corretto-11-x64-windows-jdk\jdk17.0.6_10\bin\javaw.exe
```
[image]

And if I delete the added content which is caused by changing the language, It truns back to working well.

alirana01, 2023-04-11:
@2447328758 Thanks for the quick reply we are working on the fix for this

alirana01, 2023-04-25:
@2447328758 this will be fixed with the latest release in the mean time you can open the espressif-ide.ini file and add the following lines before `-vm`

```
-nl
zh
```

This will help you start the ide in Chinese.
If the issue still persists after the latest upcoming release please feel free to reopen the issue.

closed 2023-04-25 by alirana01