espressif/idf-eclipse-plugin #757: Import IDF project from Linux to Windows Eclipse with problem

Samuel-Zhang-Rheem, 2023-04-30:
There is a Linux IDF project, it works, I try to use Windows Eclipse for debugging. But I get two issue after imported to Eclipse IDF.
product information: esp-idf-v4.4.4.

1. **Build**: It fails when I use the tool bar BUILD with this error message " _Error: Program "make" not found in PATH_".            But I can command line "idf.py build" in terminal to build the while project successfully after setup variables (export.bat).
2. **Debug**:  After setup the configuration, I press the tool bar, it generate error message "Error exist in the active configuration ...", I press the button "Proceed", then it goes to debug and run "panic.c".

Please advise how to fix this problem.
Regards
Sam

[image]
[image]
[image]

[Debug Fail Message.txt](https://github.com/espressif/idf-eclipse-plugin/files/11360807/Debug.Fail.Message.txt)
[ESP32 Error Log.log](https://github.com/espressif/idf-eclipse-plugin/files/11360808/ESP32.Error.Log.log)

AndriiFilippov, 2023-05-02:
hi @Samuel-Zhang-Rheem  ! 

Can't reproduce this issue by following steps:

step 1: Linux -> create project based on esp-idf v4.4.4 (Hello_World project) 
step 2: Import project to Eclipse (with esp-idf v4.4.4) on Windows 
step 3: build / debug - LGTM 👍 

I'm using:
Espressif IDE plugin version: Version: 2.9.1
OS - Windows 10 
Eclipse version: 23-03 
esp-idf:  v4.4.4
Could you, please, provide product information log for further investigation.

Samuel-Zhang-Rheem, 2023-05-04:
Hello,

There are some difference between example Hello_World project and my project.

I try to view the project properties, it is different between other project imported from examples.
[image]

The "Debug Fail message.txt" contains the details, it report "Error: [esp32.cpu1] Unexpected OCD_ID = 00000000", how to cause this error?

AndriiFilippov, 2023-05-05:
@Samuel-Zhang-Rheem 
Please send me product info and I will tcheck again today.  
to show product information log, go to "Espressif -> Product Information." 

Thank you !

Samuel-Zhang-Rheem, 2023-05-05:
Hello:

Here is the production information:
Operating System: windows 10
Java Runtime Version: 17.0.1+12
Eclipse Version: 4.20.0.v20210611-1600
Eclipse CDT Version: 10.3.0.202104042017
IDF Eclipse Plugin Version: 2.6.0.202208110735
ESP-IDF v4.4-dirty
Python set for IDF_PYTHON_ENV: Python 3.10.3

[image]

I talked with my colleague in Rheem USA, he said they are using VS Code to debug.
This project file structure is Eclipse, maybe someone build it with very old Eclipse/IDF tool.

I'm using Eclipse all the time, not familiar with VS code. Can I use Eclipse/IDF to build another template project, copy all those files in main directory to new project?

sigmaaa, 2023-05-11:
Hi @Samuel-Zhang-Rheem,
sorry for the late reply.
"Can I use Eclipse/IDF to build another template project, copy all those files in main directory to new project?" -Yes, it should be possible as long as the project has the correct CMake structure.
regarding your issues:
1. _Build: It fails when I use the tool bar BUILD with this error message " Error: Program "make" not found in PATH"_
- Take a look to see if the selected generator is Ninja. If it is not, change it to Ninja and see if that helps. To do this, edit your launch configuration by clicking on the gear icon.
[image]

Then click on Build Settings tab -> Generator -> Ninja

2. _Debug: After setup the configuration, I press the tool bar, it generate error message "Error exist in the active configuration ...", I press the button "Proceed", then it goes to debug and run "panic.c"._
- Maybe GDBstub on panic mode is selected in your sdkconfig file https://github.com/espressif/idf-eclipse-plugin#gdbstub-debugging. The default option there is "Print registers and reboot" 

P.S. Your IDF environment is a little bit outdated. The latest version includes important bug fixes and new features compared to version 2.6.0. If you would like to take a look at the latest version but do not want to risk breaking your environment, you can try Espressif-IDE, which has all the necessary dependencies included in a separate, lightweight IDE.

Samuel-Zhang-Rheem, 2023-06-04:
Hello Sigmaaa,

Thanks for your reply.

1. I can't change the build setting.  
[image]

2. This project has  "C/C++ build" in properties. For other working project without "C/C++ build" in properties. 
[image]

3. I build a IDF template project and copy the source code and CMake file to the template project, it can build the project, and the debugger works.

Regards
Sam

sigmaaa, 2023-06-05:
Hi @Samuel-Zhang-Rheem,

you should be able to change build settings in the launch configuration (not debug) of the project. Each project should have at least 1 launch configuration like you have `scan` and `softAp`. If you don't have the launch configuration for your project for some reason, you can try to create a new launch configuration and set the project in the main tab. 
To do this:
**1. In the launch configuration bar, click on New Launch Configuration...:**

[image]

**2. Then, select ESP-IDF application:** 

 [image]

**3. In the main tag, select the project:**

[image]

**4. Select this launch configuration and try to build with it.**

P.S. In 2.9.2 it will be more intuitive for the user, that the launch configuration is responsible for the build. In the main tab of the debugger configuration, it will be a section (the selection box)  that points to the launch configuration responsible for the build (instead of Build Configuration)

Samuel-Zhang-Rheem, 2023-06-05:
Hello Sigmaaa;

1. It report error message.
[image]

2. I can make the setting same as other projects, when I build, it report error message.
[image]

3.This is Eclipse IDF project, it maybe created by old version of IDF plugin. If import old IDF project, how to make sure it can be compile and build with new IDF plugin?

4. After I make it pass compiler, when I start OpenOCD debug, it get panic at: ret = nvs_flash_init();      Because I use EVK board instead of the application board, maybe the NVS flash is different, where can I configure the NVS?
5. My Plugin is version: 2.8.1.202302071030  If I install new version of IDF plugin, I should install in another the latest version of Eclipse in another folder?

sigmaaa, 2023-06-05:
Hi @Samuel-Zhang-Rheem,

1. I'm confused how did you get these variables in the launch configuration (${IDF_PY}, ${IDF_PYTHON_ENV_PATH}? We added them to the master like 2 weeks ago, how could you get them on version: 2.8.1.202302071030? Have you installed also the plugin from the master version and exported the launch configuration from there? If yes, you have to change it, because on 2.8.1 they will not work.

2. Can you export the log with this error, please? https://github.com/espressif/idf-eclipse-plugin#error-log With it we could get more information about the exception and will be better understand why you getting this

3. In my understanding, there should be no problem with importing the project, which was created from an old Eclipse plugin. I'll check more on this

4. to use different NVS, you have to flash it with the command https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#write-binary-data-to-flash-write-flash
You can do it in eclipse, by opening esp-idf terminal. Here is example https://www.youtube.com/watch?v=Z5Yz3g-ZSlM&t=467s. In this video you can also take a look on Partition Table editor and NVS table editor features, that can be usefull in your case if you are using custom nvs tables.
P.S. I think we can improve this workflow and add flash command to the NVS partition editor.
5. I'm not sure if I understood the question correctly. You don't have to install the latest Eclipse (also, the latest plugin version was tested on Eclipse IDE for C/C++ Developers 2022-09 and 2022-12, however older versions probably works as well).

Samuel-Zhang-Rheem, 2023-06-21:
Hello Sigmaaa,

The NVS issue fixed, because the memory map was wrong.

I use Eclipse to do other ARM MCU debugging, I can reset during debugging. But I can't do reset for Eclipse IDF,
I have to stop and download again when I try to start whole program from beginning, this will take much longer time.

Is it possible to "reset"?

sigmaaa, 2023-06-21:
Hi @Samuel-Zhang-Rheem,

you can try this:
When debugging, open the Debug view (by default in the Project Explorer section) -> Right click on some thread you want to reset, for example, the main thread -> click Restart:
[image]

sigmaaa, 2023-06-21:
the second option could be turning off "Flash every time with application binaries" in the 'Startup' tab of the debug configuration but it takes more time and requires additional manual flashing when you do want to make some changes