espressif/idf-eclipse-plugin #760: [Question] Eclipse support for ESP32-C6

GTCLive, 2023-05-09:
Hi. We've gotten a few custom boards made to test the performance of new comm protocols. So far so good, we're likely to be upgrading two or more of our previous projects which were based on the C3 and S3, over to a ESP32-C6 silicon. 

**Question:** Could soon be a good time to officially integrate the C6 toolchain into Eclipse?

ps: I've attempted to do this manually following unofficial guidance I've found within a thread here, however made more damage than progress, and so reverted this back to default. Current situation is we are forced to switch between CLI and Eclipse constantly; not a problem now during testing, but may become one productivity-wise when we're ready to develop.

sigmaaa, 2023-05-09:
Hi @GTCLive,

thank you for your question. We have already added support for the esp32-c6 target in the master branch. It will be added to the next release version, but you can try it now with this[ update site](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753024), or Espressif IDE version:

[espressif-ide-win32](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753028)
[espressif-ide-macosx.cocoa.aarch64](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753026)
[espressif-ide-macosx-cocoa-x86_64](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753027)
[espressif-ide-linux](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753025)

GTCLive, 2023-05-10:
Thank you so much, @sigmaaa. I've used the ([ update site](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753024)) and it worked first try. Everything shows up, compiles, updates, it's flawless. 
Much appreciated. 

Leaving this open for others to enjoy your generosity and kindness. Feel free to close once released, if I forget.

Nistp, 2023-06-07:
@sigmaaa hi, I'm looking to try out the esp32-c6 in the Espressif-IDE. How would I install the update provided thru the update website?

kolipakakondal, 2023-06-07:
Hi @Nistp 

Please update to the latest nightly build and this would bring esp32-c6 support. Here is the update site url.
https://dl.espressif.com/dl/idf-eclipse-plugin/updates/nightly/

How to update?
https://github.com/espressif/idf-eclipse-plugin#installing-idf-plugin-using-update-site-url

I hope that helps!

Nistp, 2023-06-07:
@kolipakakondal hi, thank you for your comment and for the help! I was able to install the nightly build and now I can build and flash esp32c6 stuff onto the esp32c6-devkitc-1v12. 

I want to mention, however, I had to explicitly install the nightly build, as when I tried updating it, it only updated to the latest and couldn't find the nightly build; additionally, I had to disable the CDT software update site, since it was causing me to get an error. Also, the welcome page doesn't load anymore  - not critical, but just as an FYI.

GTCLive, 2023-06-07:
[unofficial] Another way to do it is use the "Archive" install. The way I've done it and all went flawless. Attaching a visual for reference. **Eclipse >> Help >> Install New Software >> Click "Add" >> Select "Archive"**. And browse for the  ([update site](https://github.com/espressif/idf-eclipse-plugin/suites/12338850827/artifacts/654753024)]) file originally given by @sigmaaa.

[image]

Nistp, 2023-06-07:
@GTCLive yeah, this is very similar to what I did:
[image]

GTCLive, 2023-06-07:
@Nistp I really can't comment neither advise on the best method. I am not Espressif staff/dev/team. Just sharing a method that works perfectly fine, and which does not replicate the errors you've reported. 

Eclipse IDE for C/C++ Developers
Version: 2022-09 (4.25.0)
Build id: 20220908-1902

Nistp, 2023-06-07:
@GTCLive oh, no-no, I'm leaving my comments so anyone who is trying to do the same thing from Espressif-IDE will be able to folly my steps, just in case. I'm mentioning my errors for both end-users and Espressif staff/dev-team, just in case these errors/bugs show up for someone else.

kolipakakondal, 2023-06-07:
> esp32c6-devkitc-1v12.

Thanks for the update @Nistp 

Could you share us the IDE [Product Information log](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-do-i-view-product-information), just to see what is the base version of Eclipse CDT which you're using so that we check if we can avoid the update errors.

Nistp, 2023-06-08:
@kolipakakondal sure! Here is my IDE's Product Information log:

Operating System: windows 10
Java Runtime Version: 17.0.6+10-LTS
Eclipse Version: 4.25.0.v20220831-1800
Eclipse CDT Version: 10.7.1.202208160035
IDF Eclipse Plugin Version: 2.10.0.202306060331
ESP-IDF v5.2-dev-921-gdc016f5987-dirty
Python set for IDF_PYTHON_ENV: Python 3.11.2

Is that enough or do you need more info?

kolipakakondal, 2023-06-13:
IEP v2.10.0 is released https://github.com/espressif/idf-eclipse-plugin/releases/tag/v2.10.0