espressif/idf-eclipse-plugin #777: getting 13 semantic error while building "Hello World" in idf-eclipse plugin in ubuntu.

Taimoor852, 2023-06-02:
**Expected behavior**
while building the example project "Hello world" in idf eclipse plugin getting 13 semantic errors and have to us quick fix everytime to flash. 

**Operating System**: ubuntu 20.04
 **Java 17 and above**:  JDK Development Kit 20 
**Eclipse** : Eclipse Installer 2023‑03 
**ESP-IDF** : esp-idf-v5.0.2

**Errors**:
Description	Resource	Path	Location	Type
Field 'cores' could not be resolved	hello_world_main.c	/hello_world/main	line 25	Semantic Error
Field 'features' could not be resolved	hello_world_main.c	/hello_world/main	line 26	Semantic Error
Field 'features' could not be resolved	hello_world_main.c	/hello_world/main	line 27	Semantic Error
Field 'revision' could not be resolved	hello_world_main.c	/hello_world/main	line 29	Semantic Error
Field 'revision' could not be resolved	hello_world_main.c	/hello_world/main	line 30	Semantic Error
Symbol 'CHIP_FEATURE_BLE' could not be resolved	hello_world_main.c	/hello_world/main	line 27	Semantic Error
Symbol 'CHIP_FEATURE_BT' could not be resolved	hello_world_main.c	/hello_world/main	line 26	Semantic Error
Symbol 'ESP_OK' could not be resolved	hello_world_main.c	/hello_world/main	line 32	Semantic Error
Symbol 'NULL' could not be resolved	hello_world_main.c	/hello_world/main	line 32	Semantic Error
Symbol 'portTICK_PERIOD_MS' could not be resolved	hello_world_main.c	/hello_world/main	line 44	Semantic Error
Symbol 'stdout' could not be resolved	hello_world_main.c	/hello_world/main	line 47	Semantic Error
Type 'esp_chip_info_t' could not be resolved	hello_world_main.c	/hello_world/main	line 20	Semantic Error
Type 'uint32_t' could not be resolved	hello_world_main.c	/hello_world/main	line 21	Semantic Error

**Eclipse Error log:** 
[error.log](https://github.com/espressif/idf-eclipse-plugin/files/11634253/error.log)

**Screenshots**

[image]

**Espressif-IDE Product Information:**

Operating System: linux
Java Runtime Version: 20.0.1+9-29
Eclipse Version: 4.27.0.v20230302-0300
Eclipse CDT Version: 11.1.0.202212091724
IDF Eclipse Plugin Version: 2.9.1.202304060814
ESP-IDF v5.0.2
Python set for IDF_PYTHON_ENV: Python 3.8.10

sigmaaa, 2023-06-02:
Hi @Taimoor852,

Thank you for reporting this. Please try `right click on Project > C/C++ Index > Rebuild` it should reduce the number of indexer errors. 
Also, it's unfortunately common behavior on ESP-IDF 5.0 and higher to have some unresolved error with some imports, check [this](https://github.com/espressif/idf-eclipse-plugin/blob/master/FAQ.md#how-to-disable-the-unresolved-symbol-error-markers) while we work on a fix.

Taimoor852, 2023-06-02:
> Hi @Taimoor852,
>
> Thank you for reporting this. Please try `right click on Project > C/C++ Index > Rebuild` it should reduce the number of indexer errors. Also, it's unfortunately common behavior on [...]

Hi @sigmaaa Thanks for the reply. i will try as you mentioned in above and let you know. 

one more thing i wanted to make clear, while ESP-IDF Configuration,  i have seen something in the console 
**Configuration.Unable to copy rules for OpenOCD to system directory, try running the eclipse with sudo command**
[image]
please let me know if that will cause some problems ?

sigmaaa, 2023-06-03:
Hi @Taimoor852,

 contrib/60-openocd.rules file is needed for USB devices on GNU/Linux, so theoretically you can have some problems with debugging using openocd. 
 So I guess try running Eclipse with the sudo command or copy this file https://github.com/arduino/OpenOCD/blob/master/contrib/60-openocd.rules to the /etc/udev/rules.d/ manually